brocaar/chirpstack-network-server #204: how to change rx2 frequncy

Ankitjd18, 2017-07-14:
Hi,
I can see an option to change band, but not of rx2 frequency. Please help me here.

brocaar, 2017-07-14:
Each band has a default RX2 frequency. You could use the `RXParamSetupReq` mac-command to re-configure the RX2 frequency. See section 5.4 of the specs:

> 17 The RXParamSetupReq command allows a change to the frequency and the data rate set
> 18 for the second receive window (RX2) following each uplink.

This is not a feature that is available in LoRa Server (yet), but you could schedule such mac-command using the LoRa Server api. For an example see: https://docs.loraserver.io/loraserver/integrate/network-controller/

closed 2017-07-25 by brocaar

Ankitjd18, 2017-08-18:
I apologise on reopening this issue, I was caught up in some other issue and didn't have proper time to check.
I saw the example shared by you and also visited some previous issue related to sending of Mac commands.
In the example shared, I am clueless on construct the mac-command part, I have searched a lot in the docs and the project about the available mac commands and their sysntax but unable to find anything. Can you please help me here.
Sorry again for writing so late.