brocaar/chirpstack-network-server #205: ADR: tx power change

mogyoros, 2017-07-14:
Dear @brocaar 
I am currently evaluating the ADR functionality of loraserver.
The node is an RN2483, and ADR is set on.
On the backend the node is registered with OTAA and ADR interval=2.

After the node is reset and join successfully happens, the network server starts to change the tx power level of the node. This starts from req_tx_power_idx=1, which corresponds to 14dBm, and it decreases the req_tx_power_idx by 1 after each node transmission. So far so good. The node acknowledges these adr commands until req_tx_power_idx<5.

> Jul 14 15:24:19 nwsrv loraserver[4229]: time="2017-07-14T15:24:19+02:00" level=info msg="mac-command block added to queue" cid=LinkADRReq dev_eui=0004a30b001bc49a frmpayload=false
> **Jul 14 15:24:19 nwsrv loraserver[4229]: time="2017-07-14T15:24:19+02:00" level=info msg="adr request added to mac-command queue" dev_eui=0004a30b001bc49a dr=5 nb_trans=1 req_dr=5 req_nb_trans=1 req_tx_power_idx=5 tx_power=4**
> Jul 14 15:24:19 nwsrv loraserver[4229]: time="2017-07-14T15:24:19+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a
> Jul 14 15:24:19 nwsrv loraserver[4229]: time="2017-07-14T15:24:19+02:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/5ccf7fffff1e22fe/tx"
> Jul 14 15:24:19 nwsrv loraserver[4229]: time="2017-07-14T15:24:19+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a
> Jul 14 15:24:19 nwsrv loraserver[4229]: time="2017-07-14T15:24:19+02:00" level=info msg="pending mac-command block set" cid=LinkADRReq commands=1 dev_eui=0004a30b001bc49a frm_payload=false
> Jul 14 15:24:19 nwsrv loraserver[4229]: time="2017-07-14T15:24:19+02:00" level=info msg="mac-command block removed from queue" cid=LinkADRReq dev_eui=0004a30b001bc49a
> Jul 14 15:24:53 nwsrv loraserver[4229]: time="2017-07-14T15:24:53+02:00" level=info msg="packet(s) collected" dev_eui=0004a30b001bc49a gw_count=1 gw_macs=5ccf7ffffff7a009 mtype=ConfirmedDataUp
> Jul 14 15:24:53 nwsrv loraserver[4229]: time="2017-07-14T15:24:53+02:00" level=info msg="rx info sent to network-controller" dev_eui=0004a30b001bc49a
> Jul 14 15:24:53 nwsrv loraserver[4229]: time="2017-07-14T15:24:53+02:00" level=info msg="mac-command block removed from pending" cid=LinkADRReq dev_eui=0004a30b001bc49a
> **Jul 14 15:24:53 nwsrv loraserver[4229]: time="2017-07-14T15:24:53+02:00" level=info msg="link_adr request acknowledged" dev_eui=0004a30b001bc49a dr=5 enabled_channels=[0 1 2] nb_trans=1 tx_power_idx=5**
> Jul 14 15:24:53 nwsrv loraserver[4229]: time="2017-07-14T15:24:53+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a
> 
But the network server does not stop to decrease, so it sends again with req_tx_power_idx=6. This can not be handled by the node, since this parameter can have 1...5 value according to the RN2483 command reference and the LoRaWAN specification 1.0, section 7.1.3.

> Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="mac-command block added to queue" cid=LinkADRReq dev_eui=0004a30b001bc49a frmpayload=false
> **Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="adr request added to mac-command queue" dev_eui=0004a30b001bc49a dr=5 nb_trans=1 req_dr=5 req_nb_trans=1 req_tx_power_idx=6 tx_power=5**
> Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a
> Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/5ccf7fffff1e22fe/tx"
> Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a
> Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="pending mac-command block set" cid=LinkADRReq commands=1 dev_eui=0004a30b001bc49a frm_payload=false
> Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="mac-command block removed from queue" cid=LinkADRReq dev_eui=0004a30b001bc49a
> Jul 14 15:25:26 nwsrv loraserver[4229]: time="2017-07-14T15:25:26+02:00" level=info msg="packet(s) collected" dev_eui=0004a30b001bc49a gw_count=1 gw_macs=5ccf7ffffff7a009 mtype=ConfirmedDataUp
> Jul 14 15:25:26 nwsrv loraserver[4229]: time="2017-07-14T15:25:26+02:00" level=info msg="rx info sent to network-controller" dev_eui=0004a30b001bc49a
> Jul 14 15:25:26 nwsrv loraserver[4229]: time="2017-07-14T15:25:26+02:00" level=info msg="mac-command block removed from pending" cid=LinkADRReq dev_eui=0004a30b001bc49a
> **Jul 14 15:25:26 nwsrv loraserver[4229]: time="2017-07-14T15:25:26+02:00" level=warning msg="link_adr request not acknowledged" channel_mask_ack=true data_rate_ack=true dev_eui=0004a30b001bc49a power_ack=false**
> Jul 14 15:25:26 nwsrv loraserver[4229]: time="2017-07-14T15:25:26+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a
> 
From this point onward the network server always tries to set req_tx_power_idx=6, but the node always denies to perform this action as per above.
And I think this has an effect on the data_rate setting aswell, beacuse the node is getting commanded to set the date_rate=5 but remains on the original data rate .

>  Jul 14 16:53:22 nwsrv loraserver[4229]: time="2017-07-14T16:53:22+02:00" level=info msg="mac-command block added to queue" cid=LinkADRReq dev_eui=0004a30b001bc49a frmpayload=false
> Jul 14 16:53:22 nwsrv loraserver[4229]: time="2017-07-14T16:53:22+02:00" level=info msg="adr request added to mac-command queue" dev_eui=0004a30b001bc49a dr=3 nb_trans=1 req_dr=5 req_nb_trans=1 req_tx_power_idx=6 tx_power=5
> Jul 14 16:53:22 nwsrv loraserver[4229]: time="2017-07-14T16:53:22+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a
> 

> Jul 14 16:55:46 nwsrv loraserver[4229]: time="2017-07-14T16:55:46+02:00" level=info msg="mac-command block removed from pending" cid=LinkADRReq dev_eui=0004a30b001bc49a
> Jul 14 16:55:46 nwsrv loraserver[4229]: time="2017-07-14T16:55:46+02:00" level=warning msg="link_adr request not acknowledged" channel_mask_ack=true data_rate_ack=true dev_eui=0004a30b001bc49a power_ack=false
> Jul 14 16:55:46 nwsrv loraserver[4229]: time="2017-07-14T16:55:46+02:00" level=info msg="node-session saved" dev_addr=06ab9849 dev_eui=0004a30b001bc49a

Which results in that in time the node reaches dr=0, and unable to get back to dr=5, just with a reset.
(the gateway and node distance in this case is a couple of meter)

Could you please look into this?
I am open to test a new public release if solution is found.
Thanks.
Regards,
Tamas

brocaar, 2017-07-15:
Hi Tamas, the new LoRaWAN Regional Parameters spec defines more TXPower indices than the current RN2483 node supports, so it is expected that these nodes will nACK the tx-power change to `6`. Please note that the ADR engine first tries to increase the data-rate, before it tries to decrease the TX Power. 

> Jul 14 15:25:05 nwsrv loraserver[4229]: time="2017-07-14T15:25:05+02:00" level=info msg="adr request added to mac-command queue" dev_eui=0004a30b001bc49a dr=5 nb_trans=1 req_dr=5 req_nb_trans=1 req_tx_power_idx=6 tx_power=5

Please note that at this time, the node is already operating on `dr=5` and `tx_power=5`.

mogyoros, 2017-07-16:
Hi Brocaar!
Thank you very much for your information. I understand that additional tx power idx has been added (read in the other issue topic), and that is refused by the RN2483 firmware. But what I am trying to describe, is that in a situation where an unsupported tx power idx is sent to the RN2483 it is not able to perform dr change. 
I briefly summarize the situation. 

1. After node reset the node starts with dr=5, tx power=14. The OTAA join successfully done.

> Jul 16 15:40:27 nwsrv loraserver[4229]: time="2017-07-16T15:40:27+02:00" level=info msg="backend/gateway: rx packet received"
> Jul 16 15:40:28 nwsrv loraserver[4229]: time="2017-07-16T15:40:28+02:00" level=info msg="packet(s) collected" dev_eui=0004a30b001bc49a gw_count=1 gw_macs=5ccf7fffff1e22fe mtype=JoinRequest
> Jul 16 15:40:28 nwsrv loraserver[4229]: time="2017-07-16T15:40:28+02:00" level=info msg="node-session saved" dev_addr=06d09b52 dev_eui=0004a30b001bc49a
> Jul 16 15:40:28 nwsrv loraserver[4229]: time="2017-07-16T15:40:28+02:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/5ccf7fffff1e22fe/tx"

RN2483 state:
```
mac join otaa
ok
accepted
mac get dr
5
radio get pwr
14

```
2. packet 1 - OK
3. packet 2 - OK (note: ADR interval is set to 2)
4. packet 3 - OK. Network server instructs node to change pwr level idx.

> Jul 16 15:46:30 nwsrv loraserver[4229]: time="2017-07-16T15:46:30+02:00" level=info msg="backend/gateway: rx packet received"
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="packet(s) collected" dev_eui=0004a30b001bc49a gw_count=1 gw_macs=5ccf7fffff1e22fe mtype=ConfirmedDataUp
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="rx info sent to network-controller" dev_eui=0004a30b001bc49a
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="mac-command block added to queue" cid=LinkADRReq dev_eui=0004a30b001bc49a frmpayload=false
> **Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="adr request added to mac-command queue" dev_eui=0004a30b001bc49a dr=5 nb_trans=0 req_dr=5 req_nb_trans=1 req_tx_power_idx=1 tx_power=0**
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="node-session saved" dev_addr=06d09b52 dev_eui=0004a30b001bc49a
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/5ccf7fffff1e22fe/tx"
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="node-session saved" dev_addr=06d09b52 dev_eui=0004a30b001bc49a
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="pending mac-command block set" cid=LinkADRReq commands=1 dev_eui=0004a30b001bc49a frm_payload=false
> Jul 16 15:46:31 nwsrv loraserver[4229]: time="2017-07-16T15:46:31+02:00" level=info msg="mac-command block removed from queue" cid=LinkADRReq dev_eui=0004a30b001bc49a

RN2483 state:
```
mac get dr
5
radio get pwr
14
```

5. packet 4 - OK. Node ack's ADR request.
RN2483 state:
```
mac get dr
5
radio get pwr
14
```

6. packet 5 - OK. Network server instructs node to increase power idx=2.
RN2483 state:
```
mac get dr
5
radio get pwr
11
```

6. packet 5 - Couple of unsuccessful transmit attempts. Node dr: 5-> 2. Mac error.
7. packet 6 - OK. No action on network server side to change MAC
8. packet 7 - OK. Previous MAC command is processed. Node dr: 2->5. Network server adds MAC command to change dr 2->5 and pwr_idx=3
9. packet 8 - OK. Node ack ADR change.
RN2483 state:
```
mac get dr
[... 57 lines cut ...]
Jul 16 16:16:02 nwsrv loraserver[4229]: time="2017-07-16T16:16:02+02:00" level=info msg="node-session saved" dev_addr=072e0c46 dev_eui=0104a30b001bc49a
Jul 16 16:16:05 nwsrv loraserver[4229]: time="2017-07-16T16:16:05+02:00" level=info msg="backend/gateway: rx packet received"
Jul 16 16:16:05 nwsrv loraserver[4229]: time="2017-07-16T16:16:05+02:00" level=info msg="packet(s) collected" dev_eui=0004a30b001bc49a gw_count=1 gw_macs=240ac4ffff035c30 mtype=ConfirmedDataUp
Jul 16 16:16:05 nwsrv loraserver[4229]: time="2017-07-16T16:16:05+02:00" level=info msg="rx info sent to network-controller" dev_eui=0004a30b001bc49a
Jul 16 16:16:05 nwsrv loraserver[4229]: time="2017-07-16T16:16:05+02:00" level=info msg="mac-command block removed from pending" cid=LinkADRReq dev_eui=0004a30b001bc49a
**Jul 16 16:16:05 nwsrv loraserver[4229]: time="2017-07-16T16:16:05+02:00" level=warning msg="link_adr request not acknowledged" channel_mask_ack=true data_rate_ack=true dev_eui=0004a30b001bc49a power_ack=false**
Jul 16 16:16:05 nwsrv loraserver[4229]: time="2017-07-16T16:16:05+02:00" level=info msg="node-session saved" dev_addr=06d09b52 dev_eui=0004a30b001bc49a

RN2483 state
```
mac get dr
**4**
radio get pwr
2
```

As you can see node could not apply the previous MAC command dr: 4->5 and pwr_idx=6. And I am suspecting this is due that it gets a pwr_idx which is out of range it can set.
And for the next packets sent by the node this repeats until forever, i bad network conditions in time the node decreases further its dr until it reaches dr=0. And no chance (just with reset) to bring it back to better data rates.

Do you think it would be possible to add an option to this flow as an exception if node does not supports additional power level indexes, then the network server would not try to decrease further the power level?
Because currently we will loose the ADR flexibility at least with the RN2483 (and suppose RN2983 aswell, but i haven't checked the US version)

Thanks.
Kind regards,
Tamas

brocaar, 2017-07-17:
Thanks @mogyoros for your extended info. I understand your point now!

> Do you think it would be possible to add an option to this flow as an exception if node does not supports additional power level indexes, then the network server would not try to decrease further the power level?

What you propose sounds like a good solution. I'll look into this soon.

mogyoros, 2017-07-17:
Thanks @brocaar . Glad that I could explain the situation.
Awaiting for your feedback regarding the solution.
Of course I am ready to help and test it, when it becomes available.
Cheers,
Tamas

brocaar, 2017-08-02:
I need to do some more testing (real testing), but in the above commit I've updated the ADR engine to deal with data-rate and tx-power nACKs. In case of a nACK, it will store a maxSupportedTXPower or maxSupportedData rate which is the requested value - 1. So when TXPower 5 results in a nACK, the next time it will try with TXPower 4. The same for the data-rate.

What I still need to do:

- [ ] Implement TXPower reset in node-session in case the data-rate was changed by the node (without ADR request). Else there might be the risk that the node is using TXPower 0, where the server still things it is transmitting using TXPower 3. When LoRa Server then would ask the node to lower the TXPower by one, it would ask the node to change to 4, where 1 is desired.

brocaar, 2017-08-04:
@mogyoros All changes have been made. Would you be able to test against the `0_20_fixes` branch? https://github.com/brocaar/loraserver/tree/0_20_fixes

mogyoros, 2017-08-04:
@brocaar willing to do it, but do I understand correctly, that I need to setup a docker compose environment and compile it for myself?
Until now I've just used the precompiled binaries available for ubuntu in the repo...
If so, I need a bit more time to learn the how to... :-)

brocaar, 2017-08-04:
@mogyoros I can compile a binary for you :-) let me know what platform you're using (windows, mac, linux / 32 or 64 bit).

mogyoros, 2017-08-04:
@brocaar ubuntu linux x64 16.04 . And thanks for the shortcut! :-)

brocaar, 2017-08-04:
@mogyoros here you go: https://www.dropbox.com/sh/3a67pshdarcudrx/AABBNPK0mYsdiyxVW9bhR9Aha?dl=0

mogyoros, 2017-08-05:
@brocaar : thank you. Downloaded and checked the functionality.

> - OTAA join, node dr=5, radio pwr=14
> - 2 packets xmit ok
> - nwsrv: ADR command dr 5->5, pwr_idx 0->5
> - ADR ack, node dr=5, radio pwr=2
> - nwsrv: ADR command dr 5->5, pwr_idx 5->7
> - ADR nack, node dr=5, radio pwr=2
> - nwsrv: ADR command dr 5->5, pwr_idx 5->6
> - ADR nack, node dr=5, radio pwr=2
> - nwsrv: ADR command dr 5->5, pwr_idx 5->5
> - ADR ack, node dr=5, radio pwr=2
> 
> - Node: setting manually dr=3
> - nwsrv: ADR command dr 3->5, pwr_idx 5->5
> - ADR ack, node dr=5, radio pwr=2

I think with this procedure we can live, and the RN2483 node can accommodate its dr to the network conditions.
Cheers,
Tamas

brocaar, 2017-08-15:
@mogyoros great! I was at a conference last week, but will do a release with this updated functionality this week :-)

commit fd56e709 (2017-08-02): Update ADR engine to deal with txpower and dr nack.

Closes #205.
closed 2017-08-15 by brocaar