brocaar/chirpstack-network-server #206: Can't send multiple confirmed downlink messages in class C

iBrick, 2017-07-18:
Hi Brocaar,

Looks like I've found a bug with downlink messages for class C devices.
If I quickly send several downlink messages (5-7) in unconfirmed mode like:
mosquitto_pub -t "application/1/node/1111111111111111/tx"  -m '{"reference": "abcd", "confirmed": false, "fPort":7, "data": "aGVsbG8="}'
mosquitto_pub -t "application/1/node/1111111111111113/tx"  -m '{"reference": "abcd", "confirmed": false, "fPort":7, "data": "aGVsbG8="}'
mosquitto_pub -t "application/1/node/1111111111111113/tx"  -m '{"reference": "abcd", "confirmed": false, "fPort":7, "data": "aGVsbG8="}'
then all of them successfully arrive to node.

But if I change confirmed field to 'true':
mosquitto_pub -t "application/1/node/1111111111111113/tx"  -m '{"reference": "abcd", "confirmed": true, "fPort":7, "data": "aGVsbG8="}'
and send it quickly several times (let's say 7 times) then only 1 message arrive to node and one more message I see in the 'downlink_queue' table. Other 5 messages are lost. And second one which is in 'downlink_queue' table is never send as well.

brocaar, 2017-07-19:
Tbh, I'm not really sure if this is a bug or not. The current implementation lets you send data directly to the node, thus there is no concept of a queue in this case in LoRa App Server (not in the current implementation). You could implement a queue in your application however.

In case of confirmed data-down, the last transmitted item is kept in the database, so that on an ACK, LoRa App Server can notify you with the related `reference` of that frame. When you decide to transmit an other frame before the ACK is received, that queue item will be replaced.

The problem is that the node might not directly ACK the frame (directly). For now I leave it up to the application what to do. Wait, try to re-transmit, ... What I want to avoid is that LoRa App Server will have a never ending loop, polling the database what to transmit. This might end up in huge bulks of frames being transmitted over and over again when the node(s) do not respond. 

Note that the server should send the frame 7 times to the node in case of `confirmed: true`. It might be your node which does not expose the data as it did not send out an ACK yet.

iBrick, 2017-08-10:
Found some more info in this topic: https://github.com/brocaar/lora-app-server/issues/78
For now I close this issue. 
Regarding my node sending ACK I should check this as well. Thank you!

closed 2017-08-10 by iBrick