brocaar/chirpstack-network-server #207: change codeRate on loraWAN sensor node

amaestrini, 2017-07-19:
Dear Brocaar,
we are very interesting in your project and we are testing LoRaWAN.
We have a Lora App Server and Network server online and some gateways based on Raspberry Pi3 and a Concentrator (this one: https://wireless-solutions.de/products/radiomodules/ic880a.html) with the lora_pkt_fwd and lora-brige installed.
For the sensor node we use an Arduino UNO with a Dragino GPS/Shield (with LMIC library for LoRaWAN stack https://github.com/matthijskooijman/arduino-lmic) and this code to send data: https://github.com/dragino/Lora/blob/master/Lora%20Shield/Examples/lora_shield_ttn/lora_shield_ttn.ino

It works both in ABP and OTAA, we receive the data on the LoRa app server.
- we are able to change spreading factor from 7 to 12 in ABP mode and not in OTA. May be with OTA is not possible to change spreading factor given that is the network server that has pre-configured values?

- Can I ask you moreover if you have an idea how to change the codeRate? (now is 4/5) 

Thanks a lot
Andrea

brocaar, 2017-07-19:
> we are able to change spreading factor from 7 to 12 in ABP mode and not in OTA. May be with OTA is not possible to change spreading factor given that is the network server that has pre-configured values?

How do you change the spread-factor? In the firmware of the node itself? It might be an issue with the LMIC code.

> Can I ask you moreover if you have an idea how to change the codeRate? (now is 4/5)

LoRa Server will use the same code-rate is used for the uplink. So when you transmit an uplink with a code-rate other than 4/5, that same code-rate will be used for downlink. Please refer the documentation / source of LMIC to find out you can change this.

amaestrini, 2017-07-19:
Dear,
thanks a lot for the info.

Yes I change the spreading factor in the node code related to the LMIC
library, so it is the same for codeRate, you are right.

Thanks for your help
Andrea

-- 
*Andrea Maestrini ing.*
​Research engineer

OpenIoT
*- *Open Platforms and Enabling Technologies for the Internet of Things

​
----------------------------------------
CREATE-NET Research Center
Fondazione Bruno Kessler (FBK)
via alla Cascata 56D
38123 Povo, Trento (Italy)
F: +39 0461 31
​2424

e-mail:
​amaestrini@fbk.eu
www:
​
​ ​
*http://create-net.fbk.eu <http://create-net.fbk.eu>*

closed 2017-07-19 by brocaar