brocaar/chirpstack-network-server #208: Problem with node session creation when ENABLE_UPLINK_CHANNELS param was set

Iot-lilei, 2017-07-20:
Hi, I have a problem when not all the channels were used by the system (such as only 8 channels were used with param ENABLE_UPLINK_CHANNELS=0-7 when 96 channels could be used). However, the node session was created with EnabledChannels set to common.Band.GetUplinkChannels() which is 96. In this case, with each packet received from gateway, the loraserver would always tried to send an channel reconfig frame(LinkADRReq) because in func GetLinkADRReqPayloadsForEnabledChannels, the channel compare between node session enabled channel(96) and system enabled uplink channel(8) would always be different. I am not sure if this is the way how it works, maybe the node session should be created with  EnabledChannels set to common.Band.GetEnabledUplinkChannels() to avoid this?

brocaar, 2017-07-20:
The idea is that when your node is activated, it is configured to work on all channels. When you configure LoRa Server with `ENABLE_UPLINK_CHANNELS=0-7` it means only the first 8 channels are actually being used on the network, so LoRa Server needs to send this channel-reconfiguration frame (`LinkADRReq`) to make your node aware that it must only use these channels for uplink. The node should ACK this request after which the node-session `EnabledChannels` will be set to channels 0-7. 

In case you want to use all 96 channels, remove the `ENABLE_UPLINK_CHANNELS=0-7` configuration.

Iot-lilei, 2017-07-20:
Thank you! So it depends on the node to react with LinkADReq.

brocaar, 2017-07-20:
Yes, that is correct. You should be able to see the response of the node in the frame-logs in the web-interface. Please check if the node sends a nACK (which could also happen when the requested configuration is not supported by the node)>

closed 2017-11-19 by brocaar