brocaar/chirpstack-network-server #210: Loraserver crash when received a gateway status packet.

Iot-lilei, 2017-07-24:
Hi, I have a problem with the loraserver started with the param "GW_CREATE_ON_STATS=true", but the loraserver crash when received a gateway status packet, the log info is as follows:

7月 24 13:54:40 localhost.localdomain loraserver[4919]: time="2017-07-24T13:54:40+08:00" level=info msg="backend/gateway: gateway stats packet received" mac=ac83f36dedae8855
7月 24 13:54:40 localhost.localdomain loraserver[4919]: panic: value method gateway.GPSPoint.Value called using nil *GPSPoint pointer
7月 24 13:54:40 localhost.localdomain loraserver[4919]: goroutine 54 [running]:
7月 24 13:54:40 localhost.localdomain loraserver[4919]: panic(0x926d80, 0xc42024b2c0)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /usr/local/go/src/runtime/panic.go:500 +0x1a1
7月 24 13:54:40 localhost.localdomain loraserver[4919]: github.com/brocaar/loraserver/internal/gateway.(*GPSPoint).Value(0x0, 0x927f40, 0x0, 0xc4203516a0, 0x1)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: <autogenerated>:1 +0xd4
7月 24 13:54:40 localhost.localdomain loraserver[4919]: database/sql/driver.defaultConverter.ConvertValue(0x927f40, 0x0, 0x9b79e0, 0xc420254b20, 0x0, 0x0)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /usr/local/go/src/database/sql/driver/types.go:217 +0xd0
7月 24 13:54:40 localhost.localdomain loraserver[4919]: database/sql.driverArgs(0x0, 0xc420351b60, 0x8, 0x8, 0xc420351a01, 0x0, 0x1100000001, 0xc42030c078, 0x0)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /usr/local/go/src/database/sql/convert.go:36 +0x8cf
7月 24 13:54:40 localhost.localdomain loraserver[4919]: database/sql.(*DB).exec(0xc4200dd760, 0x9e4521, 0xc3, 0xc420351b60, 0x8, 0x8, 0xc420254a01, 0x0, 0x0, 0x0, ...)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /usr/local/go/src/database/sql/sql.go:1031 +0x3ea
7月 24 13:54:40 localhost.localdomain loraserver[4919]: database/sql.(*DB).Exec(0xc4200dd760, 0x9e4521, 0xc3, 0xc420351b60, 0x8, 0x8, 0x0, 0x0, 0x0, 0x0)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /usr/local/go/src/database/sql/sql.go:1010 +0xa2
7月 24 13:54:40 localhost.localdomain loraserver[4919]: github.com/brocaar/loraserver/internal/gateway.CreateGateway(0xc42019a840, 0xc420322100, 0x10, 0xc42000dff0)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /root/mygo/src/github.com/brocaar/loraserver/internal/gateway/gateway.go:169 +0x306
7月 24 13:54:40 localhost.localdomain loraserver[4919]: github.com/brocaar/loraserver/internal/gateway.handleStatsPacket(0xc42019a840, 0x5588aeed6df383ac, 0xed10782c9, 0x2004cae8, 0xc8bea0, 0x0, 0x0, 0x0, 0x8, 0x7, ...)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /root/mygo/src/github.com/brocaar/loraserver/internal/gateway/gateway.go:421 +0x308
7月 24 13:54:40 localhost.localdomain loraserver[4919]: github.com/brocaar/loraserver/internal/gateway.handleStatsPackets.func1(0xc420063728, 0xc420064480, 0xc42019a840, 0xc67d00, 0xc4201908c0, 0x70809, 0xc68ee0, 0xc42002a298, 0xc66740, 0xcb1b08, ...)
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /root/mygo/src/github.com/brocaar/loraserver/internal/gateway/gateway.go:383 +0xaf
7月 24 13:54:40 localhost.localdomain loraserver[4919]: created by github.com/brocaar/loraserver/internal/gateway.handleStatsPackets
7月 24 13:54:40 localhost.localdomain loraserver[4919]: /root/mygo/src/github.com/brocaar/loraserver/internal/gateway/gateway.go:386 +0x161
7月 24 13:54:40 localhost.localdomain systemd[1]: loraserver.service: main process exited, code=exited, status=2/INVALIDARGUMENT
7月 24 13:54:40 localhost.localdomain systemd[1]: Unit loraserver.service entered failed state.
7月 24 13:54:40 localhost.localdomain systemd[1]: loraserver.service failed.
7月 24 13:54:40 localhost.localdomain systemd[1]: loraserver.service holdoff time over, scheduling restart.
7月 24 13:54:40 localhost.localdomain systemd[1]: Started LoRa Server.
7月 24 13:54:40 localhost.localdomain systemd[1]: Starting LoRa Server...

It seems like something goes wrong with the GPS point. Could you help to check with that please?
The status packet is something like:
{"mac":"ac83f36dedae8855","rxPacketsReceived":6,"rxPacketsReceivedOK":6,"txPacketsReceived":4,"txPacketsEmitted":4,"time":"2017-07-24T06:16:03.112285Z"}

Regards,
Lei

brocaar, 2017-07-24:
Could you post the versions you're using?

- LoRa Gateway Bridge
- LoRa Server

Iot-lilei, 2017-07-24:
I am not using a LoRa Gateway Bridge. The LoRa Server  is built on the code from github 2017.6.19. It should be version 0.18.0. It does not always happen with a new status packet is received.

Iot-lilei, 2017-07-24:
I just retested the newest Loraserver. It seems like that if the gateway is completed new to loraserver, the stats packet must include "altitude",  "latitude" and  "longitude" information although they are optional, so the loraserver could create a new gateway in database. Or the loraserver could crash. The packet is like:
{
    "altitude": 10,               
    "latitude": 52.3740364,  
    "longitude": 4.9144401,   
    "mac": "1dee08d0b691d149",
    "rxPacketsReceived": 20,
    "rxPacketsReceivedOK": 15,
    "txPacketsReceived": 10,
    "txPacketsEmitted": 9,
    "time": "2016-06-10T14:04:53Z"
}
 If the gateway already exist in database, it is OK to send stats packet without optional information. The packet is like:
{
    "mac": "1dee08d0b691d149",
    "rxPacketsReceived": 20,
    "rxPacketsReceivedOK": 15,
    "txPacketsReceived": 10,
    "txPacketsEmitted": 9,
    "time": "2016-06-10T14:04:53Z"
}

brocaar, 2017-07-24:
I'll look into it. Could you post the output of `git describe --always` so that I know exactly from which git revision you're working?

Iot-lilei, 2017-07-24:
The info I got by command "git describe --always" is "0.19.2-4-g0cdea10".  Glad to be helpful to the program!

Regards!

brocaar, 2017-07-31:
I've fixed your issue in the https://github.com/brocaar/loraserver/tree/0_20_fixes branch (this will be released soon). See also https://github.com/brocaar/loraserver/commit/a86b28c42429beb073a7937c1211250dd9279bf5.

brocaar, 2017-08-01:
This will be addressed in the next release anyway (I'll remove the `nullable` from the db fields, as it adds more complexity than that it has any use), but I remember that this issue popped up before. It turned out that this was an issue that was fixed in Go 1.8 (https://github.com/golang/go/issues/8415).

I assume you are using Go 1.7?

brocaar, 2017-08-01:
(I've removed the bug label, as this issue does not occur in the "official" binaries compiled with Go 1.8)

Iot-lilei, 2017-08-03:
Yes, I was using Go 1.7.4. I rechecked this with Go 1.8. The bug does not happen again. I'm sorry for this.

commit a86b28c4 (2017-08-15): Remove nullable option from gw location & altitude.

Closes #210.
closed 2017-08-15 by brocaar