brocaar/chirpstack-network-server #211: mqtt mac commands

interfaith, 2017-07-24:
instead of api access to mac commands

how about simply publishing to mqtt ? to avoid api coding ?

brocaar, 2017-07-25:
The issue I see is that as an application, you want to have direct feedback: 1) the command was added to the queue **-or-** the call failed because... MQTT makes this way more complicated.

interfaith, 2017-07-25:
So in theory, the example network controller provided could be changed to
publish the battery level to mqtt ?

interfaith, 2017-07-25:
the example network-controller provided was attempted, no github found for
it.

some "imports" not arranged right ?

go run networkcontroller.go
# command-line-arguments
./networkcontroller.go:105: cannot use nsConn (type
*"google.golang.org/grpc".ClientConn)
as type *"github.com/brocaar/loraserver/vendor/google.golang.org/grpc".ClientConn
in argument to ns.NewNetworkServerClient
./networkcontroller.go:115: cannot use ncServer (type *"
google.golang.org/grpc".Server) as type *"
github.com/brocaar/loraserver/vendor/google.golang.org/grpc".Server in
argument to nc.RegisterNetworkControllerServer
./networkcontroller.go:115: cannot use &ncService (type
*NetworkControllerAPI) as type nc.NetworkControllerServer in argument to
nc.RegisterNetworkControllerServer:
*NetworkControllerAPI does not implement nc.NetworkControllerServer (wrong
type for HandleDataUpMACCommand method)
have HandleDataUpMACCommand("golang.org/x/net/context".Context,
*nc.HandleDataUpMACCommandRequest) (*nc.HandleDataUpMACCommandResponse,
error)
want HandleDataUpMACCommand("
github.com/brocaar/loraserver/vendor/golang.org/x/net/context".Context,
*nc.HandleDataUpMACCommandRequest) (*nc.HandleDataUpMACCommandResponse,
error)

interfaith, 2017-07-25:
this is better/closer
intel@jflab-gw-lora:~/gocode/src/github.com/brocaar/loranetcontrol$ go run
networkcontroller.go
networkcontroller.go:11:2: cannot find package "golang.org/x/net/context"
in any of:
/usr/lib/go-1.6/src/golang.org/x/net/context (from $GOROOT)
/home/intel/gocode/src/golang.org/x/net/context (from $GOPATH)
networkcontroller.go:12:2: cannot find package "google.golang.org/grpc" in
any of:
/usr/lib/go-1.6/src/google.golang.org/grpc (from $GOROOT)
/home/intel/gocode/src/google.golang.org/grpc (from $GOPATH)

interfaith, 2017-07-25:
this appears to run
~/gocode/src/github.com/brocaar/loraserver/loranetcontrol$ go run networkcontroller.go 
so just add mqtt publish to post the result of the mac command ?

brocaar, 2017-09-04:
Sorry for my late response, but this not going to work in case of roaming. The network-server responsible for the MAC layer might not be the network-server to which the application-server is connected to.

You could implement your own MQTT bridge for sending MAC commands, but this is not a feature that will be implemented into LoRa Server.

closed 2017-09-04 by brocaar