brocaar/chirpstack-network-server #213: Can not use the grpc client outside of loraserver

ckt1010, 2017-07-26:
Hi,
I was use code from: "https://docs.loraserver.io/loraserver/integrate/network-controller/" to build a network  controller. But I got error below:

> # command-line-arguments

> ./main.go:105: cannot use nsConn (type *"google.golang.org/grpc".ClientConn) as type *"github.com/brocaar/loraserver/vendor/google.golang.org/grpc".ClientConn in argument to ns.NewNetworkServerClient
> ./main.go:115: cannot use ncServer (type *"google.golang.org/grpc".Server) as type *"github.com/brocaar/loraserver/vendor/google.golang.org/grpc".Server in argument to nc.RegisterNetworkControllerServer
> ./main.go:115: cannot use &ncService (type *NetworkControllerAPI) as type nc.NetworkControllerServer in argument to nc.RegisterNetworkControllerServer:
>         *NetworkControllerAPI does not implement nc.NetworkControllerServer (wrong type for HandleDataUpMACCommand method)
>                 have HandleDataUpMACCommand("golang.org/x/net/context".Context, *nc.HandleDataUpMACCommandRequest) (*nc.HandleDataUpMACCommandResponse, error)
>                 want HandleDataUpMACCommand("github.com/brocaar/loraserver/vendor/golang.org/x/net/context".Context, *nc.HandleDataUpMACCommandRequest) (*nc.HandleDataUpMACCommandResponse, error)

It seems I can't use grpc to interface outside loraserver repository, becouse of this golang's "import" problem.

brocaar, 2017-07-26:
I would recommend using https://github.com/kardianos/govendor as it will flatten down vendor paths. Please open a topic at https://forum.loraserver.io/ in case you would like to further discuss the usage of govendor. I'm trying to move support questions from GitHub (and keep the GitHub issues as my backlog of features / bugs to work on).

closed 2017-07-26 by brocaar

ckt1010, 2017-07-26:
Thanks! I got it.