brocaar/chirpstack-network-server #214: Error message accuracy

laurre, 2017-07-28:
Hi, 

Firstly thanks for your work !

I got the following error : "node-session does not exist or invalid fcnt or mic". I am trying to analyze the problem, and I think the message is a bit too general and could me more precise to help me. I think 3 different errors could be raised based on the message.

I see the error is raised from internal/session/session.go. I see that other errors are raised from there, like MIC error, so does the long message really concern MIC ?

I am able to contribute, maybe you could guide me if you think this change should be made.

Thanks !

brocaar, 2017-07-28:
I'm not sure if this can be made clear, because of the following reason:

As de pool of possible `DevEUIs` is larger than the pool of possible `DevAddrs`, it is possible that the same `DevAddr` will be assigned multiple times (to different nodes).

So to find the node-session attached to a `DevAddr`, you have to go through the slice of possible `DevEUIs` and check if the MIC is correct, if so, then that node-session maps to the given `DevAddr`. So when you can't find the node-session, you don't know if this is because it doesn't exist or it was because you didn't know how to find it (using the MIC).

laurre, 2017-08-01:
OK, I think I understand the global idea.

I made some progress with my analysis. The "node-session does not exist or invalid fcnt or mic" message appears only sometimes after successful receptions. If I activate the "relax frame counter" option, I instead get a new warning that is "frame counters reset". After reading the related code in the session.go file, it seems that the frame counter up value is incorrect. What could I do to get more details ? I guess my end-device is sending a frame that is never received, isn't it ? I may have misconfigured uplink channels in my end-device.

One way I can think of to validate my point is to access the raw frame. I noticed that in case of error, it is written in log (`data_base64="XXX"`). Could it be possible to display this value in all cases ? I could see what is different from the last succesful frame from the first failing frame. I think it is useful anyway for debugging purposes.

brocaar, 2017-08-01:
This is probably related to your node resetting its frame-counters (maybe due to a power-cycle?)

closed 2017-08-01 by brocaar