brocaar/chirpstack-network-server #218: packet types

akieni, 2017-08-08:
Hi we like to establish mysql server connection to remote lora devices and we are curious if this packet forwarder can  be used for that scenario..
Thanks
chris

### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2017-08-10:
Could you tell a bit more about your usecase? E.g. is your plan to log output from your node?

akieni, 2017-08-12:
| Server 2 |------|Lora device |*********** radio link**************** |Lora Gateway |------| Mysql serve1|

This is what we want to achieve through a lora network .... Sending data back and forth between two or multiple location where there is a remote linux server with mysql... Of course this could be done through a 4G Lte , wimax or fiber but the deployment is in a rural area where we are doing forestry preservation....

brocaar, 2017-08-15:
I think this is out of scope of the LoRa Server project, but I don't see why you can't implement this yourself. E.g. put a custom application between the packet-forwarder and your MySQL db. Optionally, you could use https://github.com/brocaar/lora-gateway-bridge.

closed 2017-08-15 by brocaar

akieni, 2017-08-19:
Unfortunately, I don't have the knowledge to build a custom application but I am ready to pay the required price if you can help me....  
Thanks very much for your reply..  :+1:

siscia, 2017-08-29:
@akieni 

Feel free to send me a message, we may discuss your needs and see if it is possible to work together :)