brocaar/chirpstack-network-server #219: Question: Band Configuration

RakshitAdmar, 2017-08-09:
I want to configure the network server to work in the Indian Band. Which file do i make changes in ?
I can see in /etc/default/loraserver config file the extra frequencies option, but in my case i want to define the band corresponding to what is allowed in India.

brocaar, 2017-08-10:
I'm planning to do a new release soon which includes the Indian band :-) These changes can already be found in https://github.com/brocaar/loraserver/tree/0_20_fixes.

closed 2017-08-10 by brocaar

RakshitAdmar, 2017-08-11:
I have a very reliable system up and running and i don't want to put it down at the moment. Is there a fix for this? Is there some configuration file i can change ?

brocaar, 2017-08-15:
The band configuration is defined in code (also since some logic is defined in functions), then in your LoRa Server configuration you define which band to use.