brocaar/chirpstack-network-server #220: Bulk Register node

grendy, 2017-08-09:
Hi, 
If i have many nodes, how to bulk register node in lora-app-server?

[image]

Thanks

siscia, 2017-08-09:
The fastest way would be using the REST API exposed.

Start doing a login and save the token.
At this point pass the token at each request in the header.

You should be able to do everything asynchronously.

If register for otaa you still need to do a join...

brocaar, 2017-08-15:
I'll close this issue as https://github.com/brocaar/lora-app-server/issues would be a better place. But as @siscia mentioned, currently the API would be the best way to do this.

closed 2017-08-15 by brocaar