brocaar/chirpstack-network-server #221: [question] The 'data' field in the MQTT topic application/1/node/123/rx

eviveraart, 2017-08-11:
Hello. First of all thanks for making this software. I had some problems installing it, because we use an old ubuntu version, but i got it working thanks to the comprehensive guide and some google work. I installed the lora app server 10.1, gateway-bridge 2.1.5 and the loraserver 0.19.2. I have a lorank8 gateway. 

I have a connection with the gateway and it looks like the join with the end-device worked correctly. I receive messages in the log of each of the programs and also in the web-app at the gatway. But the web-app doesn't show anything under frame logs. When i subscribe to the mqtt topic application/1/node/123/rx (123 repleaced by the applicationID, found by subscribing to the topic application/#) the 'data' field in the json is the same as the encrypted data field in the things network. What could go wrong with the encryption? What do you need to answer my question? 

The data from the mqtt topic:

`application/1/node/0004a30b0019825b/rx {"applicationID":"1","applicationName":"testOne","nodeName":"testNode","devEUI":"0004a30b0019825b","rxInfo":[{"mac":"1dee0837b0dd02f0","rssi":-23,"loRaSNR":7.5,"name":"Lorank8","latitude":51.98869,"longitude":6.05546,"altitude":0}],"txInfo":{"frequency":868100000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":false,"codeRate":"4/5"},"fCnt":462,"fPort":1,"data":"IQ=="}
application/1/node/0004a30b0019825b/rx {"applicationID":"1","applicationName":"testOne","nodeName":"testNode","devEUI":"0004a30b0019825b","rxInfo":[{"mac":"1dee0837b0dd02f0","rssi":-47,"loRaSNR":8,"name":"Lorank8","latitude":51.98869,"longitude":6.05546,"altitude":0}],"txInfo":{"frequency":868100000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":false,"codeRate":"4/5"},"fCnt":463,"fPort":1,"data":"IQ=="}
application/1/node/0004a30b0019825b/rx {"applicationID":"1","applicationName":"testOne","nodeName":"testNode","devEUI":"0004a30b0019825b","rxInfo":[{"mac":"1dee0837b0dd02f0","rssi":-31,"loRaSNR":7.2,"name":"Lorank8","latitude":51.98869,"longitude":6.05546,"altitude":0}],"txInfo":{"frequency":868100000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":false,"codeRate":"4/5"},"fCnt":466,"fPort":1,"data":"IQ=="}`

The output from the gateway-bridge:

`INFO[21173] backend: publishing packet                    topic="gateway/1dee0837b0dd02f0/stats"
INFO[21173] gateway: sending udp packet to gateway        addr=145.131.135.50:45415 protocol_version=2 type=PushACK
INFO[21173] gateway: received udp packet from gateway     addr=145.131.135.50:50841 protocol_version=2 type=PullData
INFO[21173] gateway: sending udp packet to gateway        addr=145.131.135.50:50841 protocol_version=2 type=PullACK
INFO[21181] gateway: received udp packet from gateway     addr=145.131.135.50:45415 protocol_version=2 type=PushData
INFO[21181] gateway: rxpk packet received                 addr=145.131.135.50:45415 data="QEjRSwYA2gEByctOpw0=" mac=1dee0837b0dd02f0
INFO[21181] backend: publishing packet                    topic="gateway/1dee0837b0dd02f0/rx"
INFO[21181] gateway: sending udp packet to gateway        addr=145.131.135.50:45415 protocol_version=2 type=PushACK
INFO[21183] gateway: received udp packet from gateway     addr=145.131.135.50:50841 protocol_version=2 type=PullData
INFO[21183] gateway: sending udp packet to gateway        addr=145.131.135.50:50841 protocol_version=2 type=PullACK`

The output from the loraserver:

`INFO[21200] backend/gateway: rx packet received          
INFO[21200] packet(s) collected                           dev_eui=0004a30b0019825b gw_count=1 gw_macs=1dee0837b0dd02f0 mtype=UnconfirmedDataUp
INFO[21200] rx info sent to network-controller            dev_eui=0004a30b0019825b
INFO[21200] node-session saved                            dev_addr=064bd148 dev_eui=0004a30b0019825b
INFO[21212] backend/gateway: gateway stats packet received  mac=1dee0837b0dd02f0
INFO[21212] gateway updated                               mac=1dee0837b0dd02f0
INFO[21220] backend/gateway: rx packet received          
INFO[21220] packet(s) collected                           dev_eui=0004a30b0019825b gw_count=1 gw_macs=1dee0837b0dd02f0 mtype=UnconfirmedDataUp
INFO[21220] rx info sent to network-controller            dev_eui=0004a30b0019825b
INFO[21220] node-session saved                            dev_addr=064bd148 dev_eui=0004a30b0019825b
INFO[21225] backend/gateway: rx packet received          
INFO[21225] packet(s) collected                           dev_eui=0004a30b0019825b gw_count=1 gw_macs=1dee0837b0dd02f0 mtype=UnconfirmedDataUp
INFO[21225] rx info sent to network-controller            dev_eui=0004a30b0019825b
INFO[21225] node-session saved                            dev_addr=064bd148 dev_eui=0004a30b0019825b`

And the output from the application server:

`INFO[2288] handler/mqtt: publishing data-up payload      topic="application/1/node/0004a30b0019825b/rx"
INFO[2288] data-down item requested by network-server, but queue is empty  dev_eui=0004a30b0019825b
INFO[2293] handler/mqtt: publishing data-up payload      topic="application/1/node/0004a30b0019825b/rx"
INFO[2293] data-down item requested by network-server, but queue is empty  dev_eui=0004a30b0019825b
INFO[2298] handler/mqtt: publishing data-up payload      topic="application/1/node/0004a30b0019825b/rx"
INFO[2298] data-down item requested by network-server, but queue is empty  dev_eui=0004a30b0019825b`

Thanks in advance,

Evi

brocaar, 2017-08-15:
> But the web-app doesn't show anything under frame logs

See `--log-node-frames` https://docs.loraserver.io/loraserver/install/config/

> What could go wrong with the decryption? What do you need to answer my question?

Not sure if I understand what your question is, but the `data` field is not encrypted. It is base64 encoded.

eviveraart, 2017-08-16:
Ow, oops, I totally misread that! I expected the raw payload in the 'data' field and assumed it was decrypted wrong when I didn't receive that. I used an online base64 decoder and I got the expected pay-load. Thank you for your help!

closed 2017-08-19 by brocaar