brocaar/chirpstack-network-server #226: cmd/loraserver/main.go:28:2: use of internal package not allowed

y0913011011, 2017-08-18:
### Is this a bug or a feature request?
No
### What did you expect?
wnt to compile sucess
### What happened?
lchw@vm:loraserver$ make
Generating static files
Compiling source for linux 386
cmd/loraserver/main.go:28:2: use of internal package not allowed
cmd/loraserver/main.go:29:2: use of internal package not allowed
cmd/loraserver/main.go:30:2: use of internal package not allowed
cmd/loraserver/main.go:31:2: use of internal package not allowed
cmd/loraserver/main.go:32:2: use of internal package not allowed
### What version are your using?
go version go1.9rc2 linux/386
### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2017-08-18:
I can't reproduce this issue using the Go 1.9rc2.

Please make sure that you cloned the repository under `$GOPATH/src/github.com/brocaar/loraserver`. Else the compiler is not able to resolve the dependencies from the `vendor` folder.

y0913011011, 2017-08-19:
ths, when cloned  the repository under $GOPATH/src/github.com/brocaar/loraserver,it solved

closed 2017-08-19 by y0913011011