brocaar/chirpstack-network-server #227: Multicast downlink messages

sputnik-alarm, 2017-08-18:
Is it possible to send multicast downlink messages with LoRa Server?

brocaar, 2017-08-19:
Yes you can, however the multicast configuration is not part of the LoRaWAN protocol. You basically need to setup a multicast "node" in LoRa (App) Server (ABP) and configure all your nodes with this multicast configuration (DevAddr, NwkSKey, AppSKey, frame-counters etc...). Also, your nodes must be operating in Class-C.

sputnik-alarm, 2017-08-19:
How to choose a certain gateway from which multicast downlinks will be emited?

brocaar, 2017-08-20:
Currently LoRa Server will use the gateway nearest to the (multicast) Class-C device, based on the last uplink transmission.

sputnik-alarm, 2017-08-20:
Let's assume that some of my nodes containe only unicast address, and some have two addresses, unicast and multicast. They all sending uplunks always from their unicast addresses, and the network knows, near which gateway each of them placed. Network doesn't know which of nodes contain multicast address inside. Then server desides to send a multicatst downlink. But nobody still knows, near which gateways are placed nodes, able to assume multicast. So what mechanism will decide, from which gateway to send multicast message?

brocaar, 2017-08-29:
Currently the network-server only knows the "nearest" gateway of a single `DevEUI`, thus a multicast message can only be broadcasted through a single gateway. This means that for Class-C there is always an initial uplink requested before the network-server is able to use Class-C. I agree this can be improved further to allow more complex multicast scenarios :-)

borisbstyle, 2017-09-06:
@brocaar So if I understand correctly this sounds more like a hack rather than real multicast.
Would it be possible to send this message to all gateways or perhaps select the gateways where you want to send it to?

brocaar, 2017-09-06:
I agree that the above description is more a hack than a real solution :-) However a real multicast solution has not yet been implemented, this is on my backlog but there are a couple of features I'll be working on first.

Note that sending all multicast frames to all gateways is not really a sustainable solution either, nor is selecting the gateways to which a multicast frame must be sent.

Ideally, the end-user adding nodes should not know anything about the gateways in the field. A better solution would be letting the network know which nodes belong to the same multicast group so that it is able to detect by which gateways this group is covered (e.g. when these nodes would join using OTAA or send data using their non multicast DevAddr).

Although I'll not directly start working on this, I'm looking forward to your feedback! That way I can already define how this should be implemented.

borisbstyle, 2017-09-06:
You have a point there. The ideal solution would be if the node itself would subscribe to some multicast group. But creating groups in lora server could be fine and than just sending the message to all gateways where at least one device in the group has been active.

borisbstyle, 2017-09-06:
btw I would love to help contributing to the project, but never done any Go programming

maxpesce, 2017-11-04:
@brocaar Hi, what about the uplink messages? I mean, is the server still able to discriminate the uplink messages among the nodes that share the multicast configuration?

FabioDominio, 2018-04-10:
Hi Orne,

I am experiencing issues sending multicast downlink messages to a LoraWan node with multicast support.

Following the suggested procedure, I reated two nodes in the server: one "true" node with the unicast radio configuration of the physical node, and one "dummy" node, with a separate device EUI, with a separate radio configuration I only use for multicast.

Then I set in the node both its unicast and multicast radio configuration, which are held separately by the node firmware.

I finally tried to send a message to the dummy node, several times, in order to exploit the node multicast radio configuration.
The problem is, although the node correctly recognize the received message as multicast, it promptly rejects it because of a MAC command error.
According to the node firmware developers, this error is due to the fact that multicast does not support MAC commands.

I even tried to disable the adr on the node to prevent MAC commands for changing frequencies, but nothing changed.

By inspecting the logs in the server I found the field "macPayload", which in turns contains the "actual" payload I sent to the device.

Is it possible that the server always send MAC commands and is not possible to disable them? 
Or is it possible to disable MAC commands only for the multicast node?

brocaar, 2018-04-10:
Better multicast support is on my backlog :-) Until them I'm afraid I can't give you any support on this.

commit ea0ed1f9 (2018-08-28): Implement multicast support.

Closes #227.
closed 2018-08-28 by brocaar