brocaar/chirpstack-network-server #234: Schedule downlink

monkmahadeva, 2017-08-28:
This is the response that we are fetching when we achieve in a downlink . As mentioned it comes with the status 200 with an empty result set also.

[image]

Although we are trying to check a confirmation from the gateway side,Could you please tell us the right approach for this .

monkmahadeva, 2017-08-28:
Also the logs for lora-app-server show that the packet is en queued. Although When try the downlink the total number of queue elements stay the same. So i guess downlink is not happening right??

[image]

monkmahadeva, 2017-08-28:
{
{
{
  "id"="1",
   "devEUI"="......"
    "pending" = "false"
     ......
     ......
}
}
}
What does the pending parameter in the JSON body of api https://localhost:8080/api/nodes/[MAC]/queue mean??

brocaar, 2017-08-28:
Please use https://forum.loraserver.io for support questions 👍

closed 2017-08-28 by brocaar

monkmahadeva, 2017-08-28:
https://forum.loraserver.io/t/schedule-downlink/80