brocaar/chirpstack-network-server #235: Improve JOIN mqtt message

gillespilloudwyres, 2017-08-28:
During the JOIN process, a mqtt message is published with the following informations 
- applicationID
- applicationName
- nodeName
- devEUI
- devAddr

Could it be possible to add the RxInfo block contained on rx frame with this items, on the join frame ?
- mac
- time
- rssi
- snr
- gateway
- latitude
- longitude
- altitude

[image]

brocaar, 2017-08-29:
Yes, that would be possible and I think a good improvement :-)

If you or somebody else would like to help out, that would be great! In short it would require:

- [ ] The [JoinRequestRequest](https://github.com/brocaar/loraserver/blob/master/api/as/as.proto#L60) message must be extended with `txInfo` and `rxInfo` fields + related code changes to populated fields

- [ ] The [JoinNotification](https://github.com/brocaar/lora-app-server/blob/master/internal/handler/models.go#L62) struct in LoRa App Server must be extended with `rxInfo` and `txInfo` fields + related code changes

sa-wilson, 2019-01-17:
So, I'm just taking a look at implementing this, and it appears that JoinRequestRequest has been replaced with [HandleUplinkDataRequest](https://github.com/brocaar/loraserver/blob/master/api/as/as.proto#L55), is this correct?

If so it seems that only the [JoinNotification](https://github.com/brocaar/lora-app-server/blob/master/internal/handler/models.go#L70) struct and associated code needs extending. Just verifying before I dive into this in case I've missed something.

brocaar, 2019-01-21:
@sa-wilson that would be great. Yes, the join-notification is sent when the `HandleUplinkDataRequest` contains a new security context (`DeviceActivationContext`) (meaning a device joined / re-joined).

closed 2019-09-09 by brocaar