brocaar/chirpstack-network-server #236: Will never send comfirm ACK, when send a wrong MAC command

ckt1010, 2017-08-29:
### Is this a bug or a feature request?
Bug, Maybe....
### What did you expect?
send NewChannelReq MAC command
### What happened?
when I send NewChannelReq MAC command playload only have one byte. Error log print below, when lora-server send confirm ACK to node.
> 2616:time="2017-08-28T17:01:51+08:00" level=error msg="processing rx packet error: handling downlink data for node 0000000000472100 failed: get mac-commands error: read mac-command queue error: decode mac-command block error: not enough remaining bytes" data_base64="gGyZsgeAAgACdTu0U2X5PN86liINPa2idQ=="

The problem is: it will alway print this error, when send confirm ACK, then abandon this ACK until node rejoin. So should we check it as soon as we get MAC command? if it is wrong abandon this MAC command, not whole ACK?

### What version are your using?
git hase: bcfadd93339a30f67aa93db536f30dec15d0f820
branch: master

### How can your issue be reproduced?
send an wrong payload length MAC command.

### Could you share your log output?
showed before.

brocaar, 2017-09-04:
Could you post the steps how I can reproduce this issue?

ckt1010, 2017-09-07:
I use nodejs not golang to communicate with lora-server, so process may different. But I think you can send this packet to lora-server and check if it has problem:
{"devEUI": "0000000000472100","frmPayload":true,"cid":7,"commands":[[7]]

brocaar, 2017-10-23:
Sorry for my late reply. The above Mac-command is not valid as the payload is missing. You could use:

https://godoc.org/github.com/brocaar/lorawan#MACCommand
and https://godoc.org/github.com/brocaar/lorawan#NewChannelReqPayload

To construct the mac-command and then the `MarshalBinary()` method to generate the bytes.

closed 2018-04-15 by brocaar