brocaar/chirpstack-network-server #241: In Class C server Acknowledgement handling from Class A function

Richaagarwalrn, 2017-09-06:
### Is this a bug or a feature request?
Bug
### What did you expect?
For Class C server  while sending acknowledgement of confirmed uplink frame "Immediate = true" should be done.
### What happened?
For class C server  Acknowledgement handling has done by  Class A server function 
SendDataDownResponse. It get called and because of this  "Immediate = False" inside txinfo structure.
Due to this packet forwarder considered this downlink acknowledgement frame for a Class A device.
Getting this error several times
"ERROR: Packet REJECTED, timestamp seems wrong, too much in advance "
### What version are your using?
0.19.2-4-g0cdea10
### How can your issue be reproduced?
LoRaServer Class C ABP node RX1 window
### Could you share your log output?

brocaar, 2017-10-23:
I'm not sure if this is a bug. When a Class-C device sends an uplink, it does open RX1 like a Class-A device. So the network-server should be able to use Class-A logic for confirming the uplink frame.

Do you get this error also for Class-A devices?
> "ERROR: Packet REJECTED, timestamp seems wrong, too much in advance "

brocaar, 2017-11-19:
Looking forward to your reply on my previous comment. Else I'll close this issue.

closed 2017-12-07 by brocaar