brocaar/chirpstack-network-server #242: Changing channels (chMask)

borisbstyle, 2017-09-07:
### Is this a bug or a feature request?
feature

This is basically a question weather it is possible to send mac command to all or some nodes that the default channel layout has been changed. I was not able to find this in any of the manuals?
I see this is only happening when node is reactivated and during ADR.

brocaar, 2017-09-07:
When you change the channels (`--enabled-channels`) LoRa Server will automatically issue `LinkADRReq` mac-commands. Note that this only works for the "default" channels, the channels that are by definition known to the node. E.g. in the US there are 72 channels, for EU there are 3. This works independent of ADR (although some nodes reject these mac-commands when ADR is disabled, this is incorrect behaviour).

LoRa Server will disable channels that are part of the `--extra-channels` (CFList channels), but will not re-activate them (currently) as it is unknown if the node is aware of these channels. A future feature could be that LoRa Server will send new channel-configuration to the node for the channels part of the CFList and will ask the node to activate these channels.

borisbstyle, 2017-09-07:
ok I am just trying to understand this part...
https://github.com/brocaar/lorawan

And than particularly these MAC commands. How can I use these?
`* LinkCheckReq
* LinkCheckAns
* LinkADRReq
* LinkADRAns
* DutyCycleReq
* DutyCycleAns
* RXParamSetupReq
* RXParamSetupAns
* DevStatusReq
* DevStatusAns
* NewChannelReq
* NewChannelAns
* RXTimingSetupReq
* RXTimingSetupAns
* Proprietary commands (0x80 - 0xFF) can be registered with RegisterProprietaryMACCommand`

borisbstyle, 2017-09-12:
I am going to close this issue since my actual problem has to do with versioning of nodes
https://github.com/brocaar/loraserver/issues/192

closed 2017-09-12 by borisbstyle

borisbstyle, 2017-12-20:
@brocaar Sorry for coming back to this issue, but the behaviour described above doesnt seem correct.

Here some logging:
First ADR command
[image]

Response from device (Note there is a channel mask NACK as device doesn't know the extra channels):
[image]

The next message from server. And this goes on and on. Instead of this loraserver should start programming the extra channels according to your description?:
[image]

kuripiiritus, 2018-02-13:
Currently it seems, that with join the channel list works ok.  When the ADR is sent then the chmask 0x07 is sent on, which kills the extra channels configured in the server, gateway and node. After this the node only shoots on 3 default channels.  Is there a workaround / fix for that. Would not like to use 3 channels, but not also disable ADR also.