brocaar/chirpstack-network-server #243: Endianness of AppEUI and DevEUI is different for devices using ABP or OTAA under the same application.

belussi, 2017-09-12:
### Is this a bug or a feature request?
Bug.
### What did you expect?
Data endianness must be the same across whole system.
### What happened?
I'm using few devices under the same application. Most of nodes is using ABP activation but one is set to use OTAA. All are using the same AppEUI and DevEUI pattern. It seems that OTAA devices requires different endianness for mentioned values.

### How can your issue be reproduced?
Faulty configuration:
[image]

Working configuration:
[image]

### Could you share your log output?
Lora server logs.

Incorrect DevEUI order:
```
Sep 12 12:49:57 ttn-gateway loraserver[10929]: time="2017-09-12T12:49:57+02:00" level=info msg="node-session saved" dev_addr=26011591 dev_eui=0210230450670890
Sep 12 12:49:57 ttn-gateway loraserver[10929]: time="2017-09-12T12:49:57+02:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/b827ebfffe997bd3/tx"
Sep 12 12:49:57 ttn-gateway loraserver[10929]: time="2017-09-12T12:49:57+02:00" level=info msg="node-session saved" dev_addr=26011591 dev_eui=0210230450670890
Sep 12 12:49:57 ttn-gateway loraserver[10929]: time="2017-09-12T12:49:57+02:00" level=info msg="pending mac-command block set" cid=LinkADRReq commands=1 dev_eui=0210230450670890 frm_payload=false
Sep 12 12:49:57 ttn-gateway loraserver[10929]: time="2017-09-12T12:49:57+02:00" level=info msg="mac-command block removed from queue" cid=LinkADRReq dev_eui=0210230450670890
Sep 12 12:50:18 ttn-gateway loraserver[10929]: time="2017-09-12T12:50:18+02:00" level=info msg="backend/gateway: gateway stats packet received" mac=b827ebfffe997bd3
Sep 12 12:50:18 ttn-gateway loraserver[10929]: time="2017-09-12T12:50:18+02:00" level=info msg="gateway updated" mac=b827ebfffe997bd3
Sep 12 12:50:39 ttn-gateway loraserver[10929]: time="2017-09-12T12:50:39+02:00" level=info msg="backend/gateway: rx packet received"
Sep 12 12:50:39 ttn-gateway loraserver[10929]: time="2017-09-12T12:50:39+02:00" level=info msg="packet(s) collected" dev_eui=9008675004231004 gw_count=1 gw_macs=b827ebfffe997bd3 mtype=JoinRequest
Sep 12 12:50:39 ttn-gateway loraserver[10929]: time="2017-09-12T12:50:39+02:00" level=error msg="processing rx packet error: application server join-request error: rpc error: code = Unknown desc = object does not exist" data_base64="AHCz1X7QAG34BBAjBFBnCJBHm0/DHDo="
```

Incorrect AppEUI order: 
```
Sep 12 12:54:36 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:36+02:00" level=info msg="packet(s) collected" dev_eui=0010230450670890 gw_count=1 gw_macs=b827ebfffe997bd3 mtype=ConfirmedDataUp
Sep 12 12:54:36 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:36+02:00" level=info msg="rx info sent to network-controller" dev_eui=0010230450670890
Sep 12 12:54:37 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:37+02:00" level=info msg="node-session saved" dev_addr=26011599 dev_eui=0010230450670890
Sep 12 12:54:39 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:39+02:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/b827ebfffe997bd3/tx"
Sep 12 12:54:39 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:39+02:00" level=info msg="node-session saved" dev_addr=26011599 dev_eui=0010230450670890
Sep 12 12:54:44 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:44+02:00" level=info msg="backend/gateway: rx packet received"
Sep 12 12:54:44 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:44+02:00" level=info msg="packet(s) collected" dev_eui=**9008675004231004** gw_count=1 gw_macs=b827ebfffe997bd3 mtype=JoinRequest
Sep 12 12:54:44 ttn-gateway loraserver[10929]: time="2017-09-12T12:54:44+02:00" level=error msg="processing rx packet error: application server join-request error: rpc error: code = Unknown desc = DevEUI exists, but with a different AppEUI" data_base64="AHCz1X7QAG34BBAjBFBnCJDuo3GjotA="
```

brocaar, 2017-09-12:
I don't think there is any difference at the LoRa Server side with regards to the endian-nes of DevEUIs. Please note it could also that you need to enter the DevEUI / AppEUI using a different endian-net at the side of your node:

To debug what your node is sending (in Go):

```go
var phy PHYPayload
if err := phy.UnmarshalText([]byte("AHCz1X7QAG34BBAjBFBnCJBHm0/DHDo=")); err != nil {
	panic(err)
}
phyJSON, err := phy.MarshalJSON()
if err != nil {
	panic(err)
}
fmt.Println(string(phyJSON))
```

Outputs:

```
{"mhdr":{"mType":"JoinRequest","major":"LoRaWANR1"},"macPayload":{"appEUI":"f86d00d07ed5b370","devEUI":"9008675004231004","devNonce":"9b47"},"mic":"4fc31c3a"}
```

You can replace `AHCz1X7QAG34BBAjBFBnCJBHm0/DHDo=` with any LoRaWAN PHYPayload. E.g. you can subscribe to your gateway rx topic to retrieve these base64 strings. See also: https://docs.loraserver.io/lora-gateway-bridge/use/data/.

belussi, 2017-09-12:
Theoretically, I'm using Lmic. I'll do more debugging.

belussi, 2017-09-13:
This one may be closed.

closed 2017-09-14 by brocaar