brocaar/chirpstack-network-server #246: DownlinkQueue processing blocked after user Activate APB with Uplink frame-counter ZERO

econisdvlp, 2017-09-22:
### Is this a bug or a feature request?
Bug in Application Server Web Page:  
Allow user to do Activation with "Uplink frame-counter"=0 set to ZERO value by user.
This blocks all further data-down items sending.
Data-down further items remain in DownlinkQueue forever never been sent. 

### What did you expect?
APB Activation page should not allow SUBMIT with value for "Upload frame-counter" =0 !!!
The DownlinkQueue should deliver ALL Sent Data to Nodes and not get blocked /stalled in Queue, seem if fcnt=0 all those Sent Data remain in queue and block delivery of data-down items.

### What happened?
1. Environment use ABP activated nodes. And Data from DownlinkQueue does not get delivered after reseting Uplink frame-counters fcnt=0 via activation page in certain conditions. see below steps to replicate.

2. Node "Uplink frame-counter" indicate currently 1691 already high number value (issue does not appear for low values of counter node - less than ??, seem to work with values <20 not sure max value it works with) 

3. Via LoRa App Server REST API send data into DownlinkQueue to the node
using "POST /api/nodes/{devEUI}/queue", "Uplink frame-counter" become =1

4. Via Node Activation page set "Uplink frame-counter"=0, "Downlink frame-counter"=0 and press SUBMIT button

5. Verify Activation page has received a new packet from node "Downlink frame-counter" new value incremented to  =1692

6. Via LoRa App Server REST API send data into DownlinkQueue to the node
using "POST /api/nodes/{devEUI}/queue" - all those will not be delivered !!!

7. Node does not receive anything, Activation page indicate "Uplink frame-counter"=0 counter value, the Downlink Queue  "GET /api/nodes/{devEUI}/queue" indicate packet(s) in the queue and stay there forever !!!

### What version are your using?
lora-app-server:    v 0.7.2 

### How can your issue be reproduced?
Follow steps described above. Bug consistently repeat. Downlink Queue does not empty nor sent !!!
NOTE: Reset "Uplink frame-counter" only when "Downlink frame-counter" has high value and after 1-2 Data sent before (Uplink frame-counter >0) , have tested with values >100. 

### Could you share your log output?
Yes ...Please request anything that I can assist with ...

brocaar, 2017-09-22:
Please note that when sending confirmed data-down frames and the downlink frame-counter at the network-server side (what you're seeing through the UI) and the node side are out of sync, the node will simply drop these frames.

You really have to make sure that when you activate an ABP node, you activate it with the frame-counters that are in sync with the frame-counters at the side of the node.

Also, please update your installation, the latest version is 0.12 :-)

brocaar, 2017-10-23:
I'm closing this issue as no new information has been added over the last month and I believe it has to do with frames being rejected by LoRa Server because the frame-counters are out-of-sync.

closed 2017-10-23 by brocaar

PhatHub, 2017-11-01:
Just a question...
I have a friend attempting to make downlinks through the lora-app-server RESTfully. He has a script that queues about 10 in a loop to test an end node. Sometimes there will be one that dequeues (usually while deleting from the queue), but otherwise it'll just sit in the queue and usually they will never transmit.
However, when I delete his queued packets and I place one manually via the REST API page, that packet will actually get sent and it will get dequeued.