brocaar/chirpstack-network-server #248: Not receiving mqtt msgs for ABP device

ddewaele, 2017-09-26:
First off, thx a lot for having created a project like this.

I have an OTAA enabled device configured in the portal, and I'm receiving the Frame Logs in the portal, as well as the following mqtt messages : 

```
application/2/node/0018b2000000185f/rx {"applicationID":"2","applicationName":"app2","nodeName":"adeunis-tester","devEUI":"xxxxx","rxInfo":[{"mac":"xxxxx","time":"2017-09-26T10:58:20.994563Z","rssi":-23,"loRaSNR":9.5,"name":"ixor-gateway","latitude":0,"longitude":0,"altitude":0}],"txInfo":{"frequency":868100000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":7},"adr":false,"codeRate":"4/5"},"fCnt":6,"fPort":1,"data":"xxxxxxxxxxxxxxx="}
Client mosqsub/34189-MacBook-P received PUBLISH (d0, q0, r0, m0, 'gateway/xxxxxx/tx', ... (240 bytes))
```

I also have an ABP device configured in the portal, and I'm receiving the Frame Logs there (so I guess it is properly setup with the correct DevAddr / App and Network key), yet I'm not receiving MQTT payloads for that one. The uplink / downlink counters also remain 0 (despite the fact that I see msgs coming in the frame logs)

Any idea why it is working for OTAA but not for ABP ?

brocaar, 2017-09-26:
This is probably not a bug, but a misconfiguration. Would you mind posting it at https://forum.loraserver.io? That is a better place for support questions :-)

closed 2017-09-26 by brocaar

ddewaele, 2017-09-26:
done (https://forum.loraserver.io/t/not-receiving-mqtt-msgs-for-abp-device/163). 

thx.