brocaar/chirpstack-network-server #251: error

xvjialing, 2017-09-29:
Sep 29 04:24:23 7b1b1e555691 lora-app-server[138]: time="2017-09-29T04:24:23Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8000: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8000 <nil>}"

Sep 29 04:24:23 7b1b1e555691 lora-app-server[138]: time="2017-09-29T04:24:23Z" level=error msg="gwmigrate: list gateways error (will retry): rpc error: code = Unavailable desc = grpc: the connection is unavailable"

Sep 29 01:54:42 7b1b1e555691 loraserver[180]: time="2017-09-29T01:54:42Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}"

sunblade, 2017-10-03:
The same issue - after upgrade.

brocaar, 2017-10-03:
Looks like LoRa App Server can't connect to LoRa Server or is unable to connect. Please use https://forum.loraserver.io/ for support questions.

closed 2017-10-03 by brocaar