brocaar/chirpstack-network-server #253: node-session does not exist or invalid fcnt or mic

sabrine2020, 2017-10-04:
Hello,
I am getting this error:

`level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="QD5IAR8A9SEB3ne3FE4q052VbqR94yGekKLzzg==" 
`
The error message is not  clear enoght to be able to see the real issue, could it be more informative?

I have not yet configured any node and I am getting this error, is it because the gateway is getting data from unconfigured node?

### What version are your using?
I am running the stack under docker
Found this on the logs, is it the correct version?
` msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.21.0 `

brocaar, 2017-10-04:
> I have not yet configured any node and I am getting this error, is it because the gateway is getting data from unconfigured node?

You've answered your own question ;-)

closed 2017-10-04 by brocaar

sabrine2020, 2017-10-04:
But this message is not clear, since it could cover several causes. It could be a node that is not yours, or a node that is not well configured etc.

Is it possible to split the causes of this message?

Thanks.

brocaar, 2017-10-04:
Unfortunately not. The DevAddr and MIC is used for the node-session lookup (in order to support multiple nodes to join with the same DevAddr as the DevAddr space is quite limited).

sabrine2020, 2017-10-04:
I see, thanks!