brocaar/chirpstack-network-server #254: Failed to start the concentrator

sokolbiba, 2017-10-14:
Hi everyone,
I am trying to setup a new ic880a (SPI version) gateway using a raspberry pi 3 and the latest version of the raspbian https://www.raspberrypi.org/downloads/raspbian/ (lite version). After downloading the lora_gateway and packet forwarder and compiling them I get the following error (see picture below). Do you have any idea what might be going wrong?
Thank you in advance.

[image]

brocaar, 2017-10-14:
I'm going to close this issue as it is not related to the LoRa Server codebase. Please use https://forum.loraserver.io for support questions :-) Hint: your question might already been answered: https://forum.loraserver.io/t/lora-packet-forwarder-doesnt-start/231.

closed 2017-10-14 by brocaar

maximilianschwab, 2017-10-14:
Hi @sokolbiba,

before running the packet forwarder you have to bring the iC880A Concentrator in an initial state (reset).
I made this by running  shell script running prior executing the packet forwarder.

#!/bin/bash
echo "17" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio17/direction
echo "1" > /sys/class/gpio/gpio17/value
sleep 5
echo "0" > /sys/class/gpio/gpio17/value
sleep 1
echo "0" > /sys/class/gpio/gpio17/value

After that the packet forwarder should run without any problems. You can find some more information about preparing the reset pin control in the iC880A doc under https://wireless-solutions.de/images/stories/downloads/Radio%20Modules/iC880A/iC880A-SPI_QuickStartGuide.pdf starting at the end of page 9.

sokolbiba, 2017-10-14:
Hi @maximilianschwab 
I tried waht you suggested me and I still get the same error. I do not understand why but while using the ttn packetforwarder it is performing very well.

sokolbiba, 2017-10-14:
@maximilianschwab  I was misconfiguring the reset pin. Thank you for your help :)