brocaar/chirpstack-network-server #255: Node can join the network but can't receive any data

saboohajaz, 2017-10-18:
Hi,

The node is joining the network without an issue but i cant see any any data sent by the node after joining

[image]

Bridge
[image]

loraserver
[image]

lora-app-server
[image]

mqtt
[image]

I tried the same node with Multitech gateway. Multitech can receive data. But only after i reset the node one or two times
I am using STM32 Nucleo + SX1272 shield

saboohajaz, 2017-10-18:
The pictures above are for ABP activation.
For OTAA the node was joining but can see any data after the successful join request/accept

saboohajaz, 2017-10-18:
Furthermore,
I am using 
sudo mosquitto_sub -t "#"

Somehow following is not working
mosquitto_sub -t "application/1/#" -v

brocaar, 2017-10-18:
See your first screenshot, the node is not sending any `fPort` and `frmPayload`, so LoRa Server does not send anything to LoRa App Server as there is nothing to forward. `fPort > 0` means an application payload.

closed 2017-10-18 by brocaar