brocaar/chirpstack-network-server #257: Empty CFList in JoinAccept

x0y1z2, 2017-10-25:
I use EU_863_870 frequency plan. I created channel configuration with 5 extra channels, and assigned id to my gateway. But server sends JoinAccept with empty CFList. And it sends ADR packets with chMask, where activated only 3 channels.
In channel configuration field "Enabled channels" contains "0, 1, 2", because with other numbers I have error "invalid channel (code: 3)".

brocaar, 2017-10-26:
You need to use the `--extra-frequencies` flag, see also: https://docs.loraserver.io/loraserver/install/config/

closed 2017-10-26 by brocaar

x0y1z2, 2017-10-28:
Yes, it works.
For what purposes I can use extra channels in web-interface?

brocaar, 2017-10-30:
That is when you're using https://docs.loraserver.io/lora-channel-manager/overview/, to manage your gateway channel config.