brocaar/chirpstack-network-server #260: ADR without fOpts

x0y1z2, 2017-10-25:
Server usually sended normal ADR with fOpts to node, and node used low power, set from ADR. But 2 days ago server sended ADR with empty fOpts and device began to transmit on full power.
After this packet server didn't sended ADR packets, although ADR interval exceeded.

brocaar, 2017-10-26:
Could you post a screenshot of this frame? I'm not sure what you mean with:

> server sended ADR with empty fOpts

x0y1z2, 2017-10-28:
Frame with fOpts: null
[image]
Frame with normal fOpts (first ADR frame after device reactivating)
[image]

brocaar, 2017-10-30:
> But 2 days ago server sended ADR with empty fOpts

The network-server only sends a LinkADRReq when it things it is needed to change the data-rate or tx-power of your node and on the interval. From the above screenshot I can tell if it should kick in or not, but I don't think this is a bug.

In case you would discuss this further, please create a topic at https://forum.loraserver.io :-)

closed 2017-10-30 by brocaar