brocaar/chirpstack-network-server #261: "zero items in collect set" error occurs under pressure test

yzk0281, 2017-10-26:
### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?
When I did a pressure test on loraserver (400 gateways and each gateway send a data to bridge every second ), after several hours error happened:
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QOqQAACAbgACaiEK2JvV5XD40iPPvnKZ4vOf85hziFqb/Eze+QErXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QByEAACAbgACxg3dPpVlM9L1tOD4vFZcYEsKrmYqeoQxK+ZDf5ssXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QF5XAACAbgAC+0zyNAXWULkeYQnk25h9BQV+hssSkcWlQcqd83DLXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QHmVAACAbgAC400pfkZZq3Jdek+MVP3XLu/w0qf04JgEx3TmVGymXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QOaFAACAbgACAhikyocV/n/WaTN3ArGRhjWdUwpbBfYBORsrpDbVXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QJNpAACAbgACYr8LzUXJ+ziB/VJTrW889hJw9oVdlVo3ry0NYJydXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QCd0AACAbgACJA2cKKrDEntCErtkKXkyCe12lKJdL97PPZ0MwBn4XcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QDJ6AACAbgACfKMSNRXojKFELOFd0I9268BlQ4qIcy3Cf0qTVu4rXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QDNhAACAbgACI7ar4ZLl9km84hs3yKudFGO+UJlvkdj+nJyuhu6lXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QANhAACAbgACTP1eFaSRS1ZeLplvzKp4k74Ta4dCrBAkEnBIedClXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QJdfAACAbgACnr20E+0pXicBTm2KBrIEOFGiG1h2ROTijogOvGtFXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QCaTAACAbgACHrzDDEwewkAGwC46iYh1EhGqyktStK1WucyJNmAGXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QMiJAACAbgACV/ixwUK3xh/mLSJrf5rTBEWeJFCwIuZ+UuK/K8RZXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QLViAACAbgACXUeTHd3K9RlvQRFtORmJKuBWu1zL73aX0aHIKyYNXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QBN+AACAbgAC1eSEEdJJEUAtgJWplxAvTemDI7BJNKGoyyZfqFEyXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QIVRAACAbgACquMXbVVhR0eFMuKl5y6a8VCeDKAOarWJyzEDKg4xXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QINmAACAbgACjb2xJA9u2eOFE/8d/MKbbJRZRyElJSLn96z4EBAHXcPZtw=="
time="2017-10-26T00:26:56+08:00" level=error msg="processing rx packet error: zero items in collect set" data_base64="QCycAACAbgACA3b02P3IrTHMRTsZTXZprnYOze6/O4rcxe3J6uFQXcPZtw=="
it seemed to be a problem of redis, can anybody have any solutions?

brocaar, 2017-10-26:
> it seemed to be a problem of redis, can anybody have any solutions?

Do you have any Redis log output? In the current implementation, the TTL for the deduplication key is `2 * DeduplicationDelay` (so when your deduplication delay is 200ms, the key expires in 400ms). It seems in your case, that this is not enough for unknown reasons.

You might also want to check your memory usage, system load e.d.

brocaar, 2017-12-01:
Any update on this? else I will close the issue.

closed 2017-12-07 by brocaar