brocaar/chirpstack-network-server #263: loraserver  still can receive data frame after the gateway has been deleted

dmoons, 2017-10-30:
### Is this a bug or a feature request?
Seems it is a bug.  
### What did you expect?
Unregistered/unauthorized gateways' data should be dropped directly.
### What happened?
loraserver and lora-app-server still handle node data send/forwarded from an unauthorized/unregistered/deleted gateway.
Steps:
1. Add a new gateway into the DB and can works normal.Data can be received through this gateway.
2. Later, deleted it from the DB by lora-app-web interface.
3. The data can be received still from this gateway. It can be seen at the node/raw frame log page.
### What version are your using?
The latest version from GitHub.
### How can your issue be reproduced?
It is very easy, setup a gateway and delete it later when the data path  works
### Could you share your log output?

   INFO[0816] backend/gateway: rx packet received
   INFO[0816] packet(s) collected                           dev_eui=0000000100000000 gw_count=1 gw_macs=0000b827eba2e377 mtype=ConfirmedDataUp
   INFO[0816] rx info sent to network-controller            dev_eui=0000000100000000
-> WARN[0816] get gateways for macs error: expected 1 gateways, got 0  macs=[0000b827eba2e377]
   INFO[0816] node-session saved                            dev_addr=06975145 dev_eui=0000000100000000
-> INFO[0816] backend/gateway: publishing tx packet         topic="gateway/0000b827eba2e377/tx"
   INFO[0816] node-session saved                            dev_addr=06975145 dev_eui=0000000100000000
   INFO[0830] backend/gateway: rx packet received
   INFO[0831] packet(s) collected                           dev_eui=0000000100000000 gw_count=1 gw_macs=0000b827eba2e377 mtype=ConfirmedDataUp
   INFO[0831] rx info sent to network-controller            dev_eui=0000000100000000
   WARN[0831] get gateways for macs error: expected 1 gateways, got 0  macs=[0000b827eba2e377]
   INFO[0831] node-session saved                            dev_addr=06975145 dev_eui=0000000100000000
   INFO[0831] backend/gateway: publishing tx packet         topic="gateway/0000b827eba2e377/tx"
   INFO[0831] node-session saved                            dev_addr=06975145 dev_eui=0000000100000000
   INFO[0843] backend/gateway: gateway stats packet received  mac=0000b827eba2e377
   ERRO[0843] handle stats packet error: get gateway error: gateway does not exist
   INFO[0844] backend/gateway: rx packet received
   INFO[0844] packet(s) collected                           dev_eui=0000000100000000 gw_count=1 gw_macs=0000b827eba2e377 mtype=ConfirmedDataUp
   INFO[0844] rx info sent to network-controller            dev_eui=0000000100000000
   WARN[0844] get gateways for macs error: expected 1 gateways, got 0  macs=[0000b827eba2e377]
   INFO[0844] node-session saved                            dev_addr=06975145 dev_eui=0000000100000000
   INFO[0844] backend/gateway: publishing tx packet         topic="gateway/0000b827eba2e377/tx"
   INFO[0844] node-session saved                            dev_addr=06975145 dev_eui=0000000100000000

FlyYu, 2017-10-30:
I also have this problem. And in the case I never registered gateway, the gateway UDP points to the server will automatically register.

brocaar, 2017-10-30:
This is not a bug. LoRa Server will process any gateway data that it receives from the MQTT broker. For the sake of performance, I don't think it is a good idea to check on each message if the gateway is part of the network or not. Also, this would be really easy to spoof.

Better is to implement a MQTT authentication / authorization meganism. E.g. for Mosquitto, you could use https://github.com/jpmens/mosquitto-auth-plug. 

In case you have further questions how this could be implemented, lets continue the discussion at https://forum.loraserver.io.

closed 2017-10-30 by brocaar