brocaar/chirpstack-network-server #266: Source code compilation error

Sivan-wang, 2017-11-10:
### Is this a bug or a feature request?

When I compile this error message:
 cmd / loraserver / main.go: 358: 14: undefined: migrations.AssetDir.
I found that 'migrations_gen.go' has no implementation of 'AssetDir'

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

I can annotate this statement so that I can compile it

### Could you share your log output?

brocaar, 2017-11-13:
Please create a topic at https://forum.loraserver.io as this is more a support question, it seems to be related to your local setup (else it would have broken the build pipeline).

closed 2017-11-13 by brocaar

Sivan-wang, 2017-11-17:
Thanks, I found a problem with my 'go-bindata' program version. Updated normal.