brocaar/chirpstack-network-server #272: Confirmed retransmission not acknowledged

tommas-factorylab, 2017-11-21:
### Is this a bug or a feature request?
Bug
### What did you expect?
When retransmitting a confirmed uplink, a ACK downlink is expected in either RX1 or RX2.
### What happened?
Only the first transmission is acknowledged
### What version are your using?
loraserver version 0.19.0
### How can your issue be reproduced?
Transmit a confirmed uplink and make sure that the ACK downlink is not received (putting the node in a metal drawer is a very effective method, I have found). The node will retransmit the uplink without incrementing the FCnt (as expected). The server will receive the uplink but will not acknowledge the downlink

brocaar, 2017-11-22:
This is by design. The network-server will only confirm an uplink once. Re-transmissions with the same frame-counter are ignored as else this would allow for re-play attacks. It would also give an adversary the power to drain the downlink queue of a device and / or consume all the airtime of a gateway.

tommas-factorylab, 2017-11-22:
Hi,

This is not correct behavior. Although the retransmission content has to ignored by the server to prevent replay attacks, the downlink acknowledgement must still be sent. I am searching for the location of this requirement in the specification, but have not found it yet. It is however (implicitly) specified in chapter 2.12 of the LoRa Certification EU document (version 1.5). It is behavior for which any device is tested when certifying. 

I also think that, as the node is retransmitting the uplink, it is certain that the downlink did not arrive and thus any downlink payload should be retransmitted as well. Even if this is a unconfirmed downlink message. 

The problem is that when the downlink acknowledgement is not received for the first uplink, the node will continue retransmitting and lowering its data rate accordingly. This will result in the node exhausting its time on air, battery life and operating on the lower data rates than necessary (and exhausting its time on air and battery life even more).

brocaar, 2017-11-22:
> I also think that, as the node is retransmitting the uplink, it is certain that the downlink did not arrive and thus any downlink payload should be retransmitted as well.

The network-server can't know if this is a replay attack or re-transmission. I think in LoRaWAN 1.1 this is fixed by hashing the uplink channel index, so that the network-server can detect that the frame-counter did not increment, but that the MIC is unique (never used for the same `fCnt` and is valid.

Please see `4.3.1.2` of the LoRaWAN 1.0.2

> Acknowledgements are only sent in response to the latest message received and are never retransmitted.

tommas-factorylab, 2017-11-23:
Hi,

I find it odd that the specification states that acknowledgements are never retransmitted, even though the certification document describes requirements that state such behavior is expected:
>2.12 b: Uplink retransmission: Test server intentionally does not send an acknowledge to a ConfirmedUp packet. The End Device must retransmit the same packet with the same
sequence number. The test server will acknowledge the first retransmission. The End
Device must end retransmissions once it receives the acknowledge.

In this case, it is not explicitly defined as this the situation described can occur due to a uplink failing or a downlink failing. However, the node is unable to determine if the confirmed data transmission failed due to the uplink or the downlink failing. Either way, the node is expected to continue retransmitting the same packet until a defined amount of attempts have failed or an attempt has succeeded.
>2.12 d: Downlink retransmission: Test server ignores the first ACK reply from the End Device
and retransmits with the same sequence number. The End Device must acknowledge
the retransmission.

In this case, the expected behavior is more defined but the roles are reversed. As both parties should behave the same in this regard, this is not relevant. It is clearly stated that the retransmission must be acknowledged.

I think the specification is being ambiguous in this case. The specification does not state that retransmissions are never acknowledged, only that acknowledgements are not retransmitted. I think they mean that the ACK bit is only set in one transmission (one ACK bit for every received confirmed packet) and thus are never transmitted multiple times. If however, for example, unconfirmed uplink attempts != 1, I think the ACK bit must be set on both transmission attempts. Perhaps this is something the LoRa alliance can shed a light onto.

> The devices I've been testing with did x re-transmissions and then reported to the application layer that the uplink transmission failed + incremented their frame-counter to avoid this "ACK deadlock".
This is correct but does result in the device lowering its data rate. If I remember correctly, the RN2483 for example does not restore its data rate back the the level on which it started the confirmed uplink. The node will continue operating on this lower data rate for all subsequent transmissions even though this is not required. The network server then has to reconfigure the used data rate with ADR commands. This is very inefficient and something that I think should be undesired.

I have read your article (nice finds!) and I agree with your last scenario. This is a security flaw in the protocol as it allows attackers to extract valid acknowledgements from the server and thus faking acknowledgements of messages. This is a serious security issue and hopefully is correctly addressed in version 1.1. It can also be solved by not incrementing the fcnt of acknowledgements of retransmissions (the network server keeps acknowledging with the exact same fcnt). As the node is retransmitting, it did not receive the acknowledgement and thus should consider this acknowledgement with the same fcnt as valid. However, any subsequent transmissions will require an acknowledgement with an incremented fcnt, thus making the extracted acknowledgements (all with the same fcnt) useless. This however is not allowed by the protocol at the moment, but should be implementable in version 1.0 environments.

The other examples with the RN2483 are implementation errors made by Microchip that allow replay attacks. What I am proposing with acknowledging retransmissions should not result in this behavior. A key difference that I have made in my LoRaWAN implementation is acceptance of incoming packets on a network level and on a application level:
- Packets with a invalid MIC, invalid devAddr, invalid frame format (fPort=0 with fOptsLen!=0, etc) are all rejected on the network level and thus on the application level.
- Packets with a fCnt that is greater than the max allowed gap are rejected on the network level and thus on the application level
- Packets with a fCnt that is "expected fCnt - 1" and are of confirmed message type are accepted on the network level but not on the application level. This means that the packets are treated as valid by the network layer and thus an acknowledgement is scheduled. However, the contents of the packet are not forwarded to the application and thus any commands in the packet are not processed. The example with the garden will not happen as the application controlling the irrigation system never receives the application content of the packet (this is shielded by the network layer). I also do not process the mac commands of these packets (as I already have processed them), but this should not matter as mac commands are stateless (executing a command twice does not result in different behavior). 
- Packets with a fCnt that is "expected fCnt - 1" and are of unconfirmed message type are rejected on the network level and thus on the application level.

brocaar, 2017-11-23:
I think for now I leave it as is, as this is at least in line with the LoRaWAN 1.0.2 specification. It would be good however to contact the Alliance and ask them for an explanation how this relates to the certification document.

> Packets with a fCnt that is "expected fCnt - 1" and are of confirmed message type are accepted on the network level but not on the application level.

The problem here is that an adversary still gets control over the airtime of your gateways. E.g. by performing X re-play attacks for confirmed uplink, it would mean that the network-server would be able to potentially silence your gateways for some time as they run out of their duty-cycle. Also for Class-C the network-server remembers the last gateways for uplink. An adversary would also get the power to perform a replay-attack close to gateways out of reach of your node. This would mean that the for future Class-C scheduling, the network-server would use a gateway which is out of reach of your device for downlink.

An other problem that it introduces is that the downlink frame-counter would be incremented each time, meaning that for LoRaWAN 1.0.2 devices, an adversary would be able to invalidate already scheduled frames (the encryption function takes the downlink fCnt as one of its parameters).

Currently downlink frames are encrypted on the fly by LoRa App Server, but this is going to change as I'm going to move the device-queue to LoRa Server, in preparation for Class-B scheduling.

alexanderplanar, 2017-11-29:
Hi, brocaar!
I have a similar problem.
LoRa Server prints: 
_processing rx packet error: get device-session error: device-sessiont exist or invalid fcnt or mic" data_base64="gJTcWAeCGQMDBwIQEp/kejr5yoPN5XfFZNqPOWlMYQ==_

Last time I noticed strange behaviour of my LoRa stand: sometimes uplink packets can not be confirmed by the server, and the node have to retransmit it during so long time before it can drop the packet and take another from the queue. Ofcourse, it drains the battery, and I have to find solution.
I checked the the transmitter hardware, receiver hardware, the filters, but I did not find anything. At last, I connect the node to the gateway directly using SMA-cable, but I still get errors sometimes! Then I had guess that it is not radio path fault.

I has been solving this problem long time and found out, that the problem can be replayed the following way - it is required to prevent receving the acknowlege message by LoRa Gateway (Packet Forwarder) from LoRa Server. Since the packet forwarder does not receive the packet to send, LoRa Server has grounds to reject any packets with the same FCnt from the node, but the node unable to know if the server does not respond, or the gateway has not translate the message, or the message has been corrupted in radio path. And the node have to retransmit the message several times. 
If the errors due to the radio path conditions we unable to avoid, then we have to avoid the logical errors due to the protocol.
I guess, since the ACK message is packed into UDP-packet which is sent via network, the UDP can be dropped sometimes by network equipment, and the Gateway does not receive to ACK to send, then the node does not receive the ACK and it is forced to transmit the message again.
Summary, it is not fault of neither the server or node, but the fault of the communication line between the LoRa server and Lora Gateway. This line must be more reliable many times than the radio path, but instead of this it actually introduces it's own errors into the already unreliable communication radio-channel.
And the node have to pay for it by battery and the time.

Can you offer any solution?

brocaar, 2017-11-29:
I don't think I can. This is a limitation in the protocol. However I believe that in LoRaWAN 1.1 the confirmed uplink retransmit is configured to be `NbTrans`. I think this value by default is `0` and can be controlled through ADR (the current ADR engine already calculates the packetloss and in case so, it increases the `NbTrans` value. This should limit the number of times the device re-tries to transmit.

tommas-factorylab, 2017-11-29:
Hi,
>I think for now I leave it as is, as this is at least in line with the LoRaWAN 1.0.2 specification. It would be good however to contact the Alliance and ask them for an explanation how this relates to the certification document.

That is of course entirely your decision :)
>The problem here is that an adversary still gets control over the airtime of your gateways. E.g. by performing X re-play attacks for confirmed uplink, it would mean that the network-server would be able to potentially silence your gateways for some time as they run out of their duty-cycle.

That is correct. However, you can limit the number of retransmissions to a reasonable amount (for example 8, this is compliant with chapter 18.4 of the specification).
> Also for Class-C the network-server remembers the last gateways for uplink. An adversary would also get the power to perform a replay-attack close to gateways out of reach of your node. This would [...]
That is also correct. The same applies for class B devices. However, when a node is retransmitting, you could decide not to update the class B or C downlink routes for that node. This way, the node will still remain available for class B or C downlinks. I think it would be best if the uplink retransmission is answered in the cell it is coming from. This requires more complex node management (I think).
>However I believe that in LoRaWAN 1.1 the confirmed uplink retransmit is configured to be NbTrans

This is correct. They moved the responsibility of the QoS of confirmed data packets from the node to the network server. I think this is a very curious decision as the network server has no knowledge of the QoS required (which might change over time or is different for packets on for example different ports). But this is a different discussion.
Reducing the number of retransmissions should not be considered to be a solution, I think. It should be considered to be a workaround: The impact of the incorrect/undesired behavior is more limited but the actual problem (the node attempting retransmission while the server ignores all uplinks due to the invalid fCnt) is still present. I think contacting the alliance, to clarify the desired behavior, would be a good idea. I will attempt to do so.

@alexanderplanar Aren't there packet forwarders that use TCP instead of UDP? I am however not sure if TCP will work as the retransmission of the TCP packet might be to slow (this retransmission scheme depends on frame counters, if I remember correctly). The network server only has less than half a second to schedule a response for RX1. Using RX2 instead of RX1 and/or increasing the RX delay might provide a solution for this. You could also rewrite the packet forwarder to send multiple UDP packets, thus increasing the chance that at least one will arrive.

alexanderplanar, 2017-11-30:
@brocaar, thank you for reply. As I understood, the only way is to find the balance btw the retransmissions amount and battery life. Of cource, I understand, it is a Semtech UDP protocol disadvantage. As for me, I think it is a bit strange to introduce so complicated system (SSL, MQTT, Encryption), but to make the project vulnerable by design using the unreliable delivery protocol. 
_P.S. of course, it's just grumbling, I have no complaints against you._ )))

@tommas-factorylab, thank you for reply. Unfortunately, I do not know, I did not come across such implementations yet.
I think it depends on the ACK timeout, but you are right, maybe, TCP is a little bit slowly than UDP.

brocaar, 2017-11-30:
It depends on your case if the UDP protocol is the issue. I've seen other cases where there were collisions on the downlink, causing the gateway not to send the frame or at least the device not to receive it. This is something that could be improved at the LoRa Server side (currently it is fire and forget).

alexanderplanar, 2017-11-30:
@brocaar,
Do you mean excactly the gateway unit does not emit the packet however it has received from server?

tommas-factorylab, 2017-12-01:
@alexanderplanar The gateway is only capable of transmitting one packet simultaneously but can receive over a dozen packets simultaneously. If all these packets require a confirmation, this results in a scheduling conflict at the gateway. The packetforwarder will drop all tx commands except the first due to this scheduling conflict. This behavior can be detected when in the stats of the packet forwarder, the tx packets received > tx packets emitted.
The lora server could take the limitations of the gateway into account and not issue tx commands to the gateway of packets that could never be transmitted anyway (due to the scheduling conflict). This will allow retransmissions to succeed as the ack was never issued by the lora server and thus this could not result into a possible replay attack.

@brocaar 
I have found something regarding the retransmissions in the specifications: In the recommendation segment of the specification, chapter 18.1, the behavior of confirmed data and retransmissions is described. The diagram shows that the gateway (network server) is expected to send acknowledgments on retransmissions. 

Of course, these are recommendations and I find the segment to be poorly written as it also states that the retransmission "must be done on another channel". This is not what my implementation does as a single channel configuration is valid and thus it is impossible to comply with both requirements. I also do not see any advantage of forcefully choosing another channel. Instead, I continue using a pseudo-random chosen channel. Furthermore, the diagram shown also allows extraction of valid acknowledgements and thus is vulnerable to replay attacks. It is up to you to decide what to do with these recommendations :).

brocaar, 2017-12-01:
@tommas-factorylab You're right, however this is not in line with

4.3.1.2 of the LoRaWAN 1.0.2 spec:

> Acknowledgements are only sent in response to the latest message received and are never retransmitted.

> I also do not see any advantage of forcefully choosing another channel.

Note that in LoRaWAN 1.1 the MIC hash contains the uplink channel number, so in this case the network-server will see a re-transmission with an unique MIC, so it knows it can't be a replay attack.

tommas-factorylab, 2017-12-02:
@brocaar I think that requirement of the specification must be interpreted differently. The requirement states two things:
- A acknowledgement is only sent in response to the latest message received
- A acknowledgement is never retransmitted.

The first part means that an acknowledgement can only apply to the latest message received. It is not allowed to acknowledge messages received earlier but not acknowledged yet. The following scenarios are not allowed:
- TX cnf fCnt 0 --> No RX --> TX cnf fCnt 0 --> No RX --> TX failed --> TX uncnf fCnt 1 --> RX with ACK (acknowledging the fCnt 0). 
- TX cnf fCnt 0 --> No RX --> TX cnf fCnt 0 --> No RX --> TX failed --> TX cnf fCnt 1 --> RX with ACK (acknowledging the fCnt 0). 
- TX cnf fCnt 0 --> No RX --> TX cnf fCnt 0 --> No RX --> TX failed --> TX cnf fCnt 1 with invalid MIC or invalid CRC or something else --> RX with ACK (acknowledging the fCnt 0). 

The second part that a single RX may never lead to multiple transmissions (this does not include retransmissions) with ACK set. The following scenarios are not allowed:
- TX cnf fCnt 0 --> Rx with ACK --> TX uncnf fCnt 1 --> RX with ACK (acknowledging the fCnt 0). 
- TX cnf fCnt 0 --> Rx with ACK --> Ping RX/Class C RX with ACK (acknowledging the fCnt 0). 

The specification does not explicitly state that retransmissions must not be acknowledged nor does it intend to (I think). This seems to be confirmed by the recommendations in chapter 18.4 and in the test descriptions in the certification documents. 

> Note that in LoRaWAN 1.1 the MIC hash contains the uplink channel number, so in this case the network-server will see a re-transmission with an unique MIC, so it knows it can't be a replay attack.

That is correct, but I think that this could never be used to guarantee a unique MIC and thus avoid replay attacks as this would make retransmitting on the same channel during the transmission process of a packet impossible. This would for example limit the number of transmission attempts to 3 when using the default EU868 channels and without degrading the datarate (which is not possible when already operating on the lowest data rate). This number of transmission attempts is (I think) unreasonably low.

brocaar, 2017-12-04:
> The specification does not explicitly state that retransmissions must not be acknowledged nor does it intend to (I think). 

I still think it does, because it opens an attack vector for adversaries to steal your gateway airtime. Note that a gateway can receive 8 frames simultaneously but only transmit 1 (and not receive). This makes downlinks really expensive for your network and you really want to avoid giving somebody else the power to generate downlink transmissions.

> This would for example limit the number of transmission attempts to 3 ... This number of transmission attempts is (I think) unreasonably low.

So by allowing retransmissions with non-unique MICs, you would want to give an adversary the power to let the network-server re-transmit the same ACK let's say 20 times? If the network-server would accept re-transmissions on a different data-rate using the same MIC, it would be even worse as it allows an adversary to change to SF12 which will be killing for your network. 

I think really the only way to avoid this is never send an ACK on a (seen) retransmission of a confirmed uplink unless you can really be sure that it was transmitted by **your** device. The only way to be 100% sure is by making sure the MIC is unique per re-transmission, which is what 1.1 solves by including the uplink channel in the MIC hash. 

Please let me know if you see an other way to implement this in a secure way where an adversary does not get power over your gateway airtime.

alexanderplanar, 2017-12-15:
Hi everyone :)
@tommas-factorylab, yes, I understand that multiple packets requiring ACK can cause collisions, but I have only one Node connected to server, there no LoRa packets received by gateway except  the ones sent by my Node.
I guess, the problem is the server does not respond in up time. I figured out that sometimes the server send ACK frame too late. I did not solve the problem yet.
I'm not sure I've configured delays correcly, but I'm going to test new version of LoRa server, maybe, the problem will disappear itself.

brocaar, 2018-05-10:
Is this issue still present with the latest version? Else I will close this issue.

sa-wilson, 2018-05-22:
Still there on 0.26.3.

This is causing issues for me right now as I'm working on a device for flood monitoring that needs to transmit confirmed (10 trials) if it's alarmed, and re-transmit every 5 minutes until it receives an ack, unfortunately only the first transmission is ack'ed and the device misses it occasionally as I'm testing at reasonably close range. The frame counter is also not incremented when the ack is missed, so the server never acks it again and it gets stuck with the constant re-transmit timer. Although each set of transmissions does have a unique mic, as there is a timestamp in the packet that changes, the latter ones are never acked.

It's quite possible that this could happen in the field too, and our trials for this device are going to take place using loraserver, so we need to find a way to allow for re-acking of the last frame counter. Airtime concerns aren't there for us as this will be on a private network, and we don't have any airtime/duty cycle limitations (Australia).

Perhaps you could explore the option of making this behaviour configurable for those of us who desire it? I do get that it is dangerous for public networks, and probably shouldn't be the default behaviour.

brocaar, 2018-05-22:
Please note that this behavior is in line with the LoRaWAN specification. I've checked this with people from within the Alliance and their feedback is that the confirmed-data should not be used to guarantee that a frame has been delivered.

Better options are to implement a frame-counter on application-level (so within the `FRMPayload`) or to add redundancy / error correction so that you can recover from some packet-loss. There might be a way to improve this in LoRaWAN 1.1 as the uplink channel is part of the MIC but until then I recommend the above. 

If you want to go for the insecure way, you can always disable the frame-counter check (see the device configuration in LoRa App Server).

closed 2018-05-22 by brocaar

Brulath, 2018-05-22:
> Please note that this behavior is in line with the LoRaWAN specification. I've checked this with people from within the Alliance and their feedback is that the confirmed-data should not be used to guarantee that a frame has been delivered.

If confirmed data does not guarantee a frame has been delivered in this implementation of LoRaWAN, what does it do? 

Are all devices which are reliant upon confirmed messages as acknowledgements not able to be used with loraserver in case they use the reference MAC implementation (which suffers from this issue)? 

If acknowledgements aren't received for an acknowledged packet, the reference MAC assumes it didn't go through and does not increment the frame counter. Subsequent acknowledged transmissions re-use the same frame counter, with entirely different content and MIC. It is still ignored by loraserver.

Implications of this decision:
1. Each device which uses a implementation of the MAC layer which does not increment the frame counter on acknowledged message failure may become stuck in an endless loop and become unresponsive when using loraserver (particularly with ABP)
2. Each device manufacturer with acknowledged messages must implement loraserver-specific acknowledgement mechanisms
3. Each purchaser of a device must implement manufacturer-specific acknowledgement mechanisms in their application
4. RX windows must take into account the time it takes for the target application server to generate an acknowledgement and return it, or each device must transmit repeatedly until an acknowledgement from a previous message succeeds
5. Data size of packets is now increased by 2 - 4 bytes to accommodate an additional frame counter

Not quite sure I follow the part about LoRaWAN 1.1 adding uplink frequency to the MIC improving the security; those frequencies are well-known and limited in scope, you'd still encounter this problem with ~8 transmissions in most deployments.

brocaar, 2018-05-22:
@Brulath I wish this was implemented (by the specification) in a different way. E.g. for better consistency having both an uplink and downlink frame-counter. When I discussed this, the advise was that the confirmed-data type is for increased consistency, not to guarantee consistency. The latter case should be implemented on the application layer. 

Taken from the 1.1 specification:

```
2410 If an end-device does not receive a frame with the ACK bit set in one of the two receive
2411 windows immediately following the uplink transmission it may resend the same frame with
2412 the same payload and frame counter again at least ACK_TIMEOUT seconds after the
2413 second reception window. This resend must be done on another channel and must obey the
2414 duty cycle limitation as any other normal transmission. If this time the end-device receives
2415 the ACK downlink during its first receive window, as soon as the ACK frame is demodulated,
2416 the end-device is free to transmit a new frame on a new channel.
```

By adding the channel into the MIC, the network-server can validate that the re-transmission is valid (same payload but different channel is unique MIC). An adversary would be able to try a replay-attack, but as it is not able to generate an unique valid MIC, such an attack can be detected and dropped by the network-server.

I believe to implement the LoRaWAN specification correctly, you should alway increment the frame-counter,  be it a successful or failing (re)transmission. The same frame-counter must never be used for a different payload!

Note that an implementation that would always respond to a confirmed uplink message (when the frame-counter was already seen / did not increment) could create interesting attack vectors. E.g. an adversary could replay lots of confirmed uplinks, causing your network to retransmit the ACK over and over again (potentially using all your gateway capacity).

tommas-factorylab, 2018-05-22:
@brocaar I am currently working on other projects and thus cannot test the behaviour of the new version.

@brocaar @Brulath The difference between confirmed and unconfirmed is that, if the number of transmission attempts is reached, unconfirmed assumes the packet did arrive while confirmed does not. The confirmed data mechanism in version 1.1 cannot guarantee that the data will arrive as the number of transmission attempts is controlled by the network and not by the node. In version 1.0, the number of attempts is controlled by the node (and it can set it to infinite). I think that the implementation of version 1.0 is better than version 1.1 as the desired certainty that the packet arrives is dependent on the application used. The network has no knowledge of the application, but the node does. It therefore should be in control of number the transmission attempts.

However, this would allow a (malfunctioning) node or an attacker to drain the tx time of the gateways. However, I think that the network server should manage this by, for example, only acknowledging tx on a lower data rate and at max one acknowledge per 30 seconds.

> Please note that this behavior is in line with the LoRaWAN specification. I've checked this with people from within the Alliance and their feedback is that the confirmed-data should not be used to guarantee that a frame has been delivered.

That does not mean that when the first acknowledgement is not received by the node and it therefore retransmits no new acknowledgement should be sent, does it? As that would contradict the recommendations and the certification specifications. Could you confirm this?

tommas-factorylab, 2018-05-22:
> I do not agree on this. Also the LoRaWAN 1.0.2 specification says:

That piece of the specification does not state anything relating to the number of transmission attempts. The NbTransmission field of the ADR command states that it only controls unconfirmed transmissions in version 1.0. In version 1.1 is controls both unconfirmed and confirmed data transmisssions. As the network issues these commands, the network is in charge of the confirmed tx attempts in version 1.1 but not in version 1.0. 

The only thing that piece of the specification states is that a device may not send multiple packets with the ack bit set for one received confirmed data packet. Eg, when a node receive a confirmed downlink, it may sent only one packet with the ack bit set. If it receives another confirmed downlink, it again may sent only one packet with the ack bit set. If the node sends a packet without the ack bit set, the network is to assume the downlink did not arrive. If the node does not set the ack bit, but did receive the confirmed downlink, it will fail certification. If a node sends multiple packets with the ack bit set for only one received confirmed packet, it will fail certification. If a node does not acknowledge retransmissions, it will fail certification (this happened to me when doing certification, the ack bit was only set for the first attempt, further attempts did not set the ack bit). 

> The big difference with LoRaWAN 1.0.2 and LoRaWAN 1.1.0 is that with LoRaWAN 1.0.2 you can not distinguish a rentransmission from a replay attack, with LoRaWAN 1.1.0 you can given that the retransmission was done using a different channel (and / or data-rate which is also part of the MIC).

Although it is true that attackers can retransmit valid packets in version 1.0, I do not consider it to be a valid replay attack: The frame count ensures that the application layer only receives a packet once. Attacking the network layer is possible, but this can only result in the tx time being drained. As I have stated before, other solutions are possible. In version 1.1 they have circumvented this draining vulnerability by adding the DR and channel in the MIC and adding restrictions on the channel usage. This unfortunately also significantly restricts the number of transmission attempts, especially if only the default channels are used in a EU region. For US and AUS, this is not a very significant problem. I think it is a poor trade off between functionality and security.

> I don't know the the certification specifications, but I think it is best to follow the protocol specifications here which is saying that an ACK must never be retransmitted.

I believe you are misinterpreting the concept of retransmitting acknowledgements in this context (I have described the meaning of the specification above, your interpretation contradicts two other specifications). The specification is unspoken on this subject (should a retransmission be acknowledged). Unfortunately, I cannot directly contact the LoRa alliance due to restrictions in the alliance (influence, membership status, costs, etc.) and therefore cannot get confirmation regarding this subject. I can only reference to the recommendation segment and the specifications specified in the certification document.:
Certification specification:
> 2.12 d: Downlink retransmission: Test server ignores the first ACK reply from the End Device
and retransmits with the same sequence number. The End Device must acknowledge
the retransmission.

Recommendation segment:
[image]
What I can do is test this behavior with KPN, if you think that would be helpful. KPN however, is not a very good reference as they do not even correctly check frame count...

> To me this doesn't sound very secure. E.g. by allowing an adversary to do a replay attack on a lower data-rate, it is now able to use even more airtime of your gateways by using SF12 for example. I really favor to stay with the MIC validation and frame-counter validation to keep the network secure.

I do not believe it influences the security of the network, only the quality of services as attackers could perform a denial of service attack. The solutions I mentioned above are only possible solutions. You could , for example, also limit the number of acknowledged retransmissions to 2 and then one every 5 minutes or at max 5 seconds of tx time for every confirmed packet, etc. I think that the LoRa specification should have specified a back-off mechanism (increase the time between attempts by 10 seconds). Instead, according to the specifications and recommendations devices burst attempts until the duty cycle exceeds or until chapter 7 comes into play. However, simply ignoring any retransmission is (I believe) not compliant with the specifications.

Please note that I can only use my spare hours on LoRaWAN related issues as I am currently working on different projects. I might not be able to respond in a quick manner.

brocaar, 2018-05-23:
I'll be at the LoRa Alliance All Members Meeting in two weeks and will try to get this cleared out :)

Brulath, 2018-05-24:
We just noticed this also breaks adaptive data rates in the reference implementation, causing devices to always back off to data rate 0 and resetting to enable all channels if any acknowledgement is missed.

A frame is created, then transmitted multiple times in the event of acknowledgement failure. After multiple attempts it begins to back-off, reducing data rate until a success occurs. Loraserver will never respond to these repeated transmissions, resulting in an ultimate back-off to data rate 0 and full channel reset being adopted by the device.

The only solutions we can see currently is to either disable frame counters, fork loraserver, entirely gut the reference implementation to overhaul ADR to accommodate loraserver, disable adaptive data rates, or always transmit on data rate 0 (as the designed back-off mechanisms cannot work with loraserver). They all seem sub-optimal in their own special ways.

It'll be interesting to see how the LoRa Alliance are expecting the back-off to work if retransmitted frames cannot be acknowledged.

tommas-factorylab, 2018-05-27:
@brocaar That would be great. Please let us know the outcome :).

@Brulath ADR should not be affected by this problem as ADR ignores retransmissions:
>Each time the uplink frame counter is incremented (for each new uplink, repeated transmissions do not increase the counter), the device increments an ADR_ACK_CNT counter. After ADR_ACK_LIMIT uplinks (ADR_ACK_CNT >= ADR_ACK_LIMIT) without any downlink response, it sets the ADR acknowledgment request bit (ADRACKReq).

The data rate should decrease due to the confirmed packets not being acknowledged (chapter 18.4 of version 1.0). That should not affect the channels (or did they change that in version 1.1?). Could there be a problem regarding the implementation on your node? The certification program does not check for this, if I remember correctly (Test AdrAckRec does not test with nbTransmissions >= 2, only with nbTransmissions <= 1)

Brulath, 2018-05-28:
[image]

@tommas-factorylab That's true, I made a mistake there. ADR requires unique frame counters to fail 64+32 times (in AU915) in order to swap to a lower data rate, at which time it may be reasonable to remove ADR channel masks (but that is not defined, as far as I can see). I find it a bit confusing because ADR failure conditions are aimed primarily at unconfirmed messages, where 18.4 back-off seems aimed at confirmed and conflicts in how quickly a back-off should occur.

Our implementation currently backs off both within a single confirmed message (on re-transmissions) in addition to resetting on ADR counter failure conditions. An unlikely scenario occurs if a bad ADR value is provided to the node (asymmetric signal strength); the confirmed re-transmission back-off system would allow it to recover and send data prior to the 96 transmissions required for ADR back-off, except that is not possible in loraserver because all of the lower data-rate re-transmissions will be ignored.

[image]

This section states that the last data rate used will be the data rate used to transmit. It doesn't specify whether that condition only occurs upon successful transmission or not, however. It also doesn't specify the interaction with ADR (whose recovery system seems aimed at unconfirmed messages). It does imply that if one confirmed message fails then the node is expected to transmit up to 7 times at lower data rates (all of which will also fail in loraserver).

tommas-factorylab, 2018-05-30:
@Brulath I think you are correct in that ADR seems to be useful for unconfirmed message, and not really for confirmed messages. In this context, ADR could be seen as a slower implementation of the mechanism described in chapter 18.4: Instead of decrementing the data rate every 2 (re)transmissions, the data rate is decremented every 64+32 packets. The confirmed packets mechanism provides partial automated data rate management on its own as the data rate is decreased whenever no acknowledgement is received. ADR does the same (but much slower), but also allows to increase the data rate again via ADR commands. 

>This section states that the last data rate used will be the data rate used to transmit. It doesn't specify whether that condition only occurs upon successful transmission or not, however.

My implementation uses the last used data rate irregardless of the transmission result. I think this would be essential as else the node may never be able to regain network connectivity (if the node is operating on data rate 5 and performs 8 transmission attempts, it will never reach data rate 0, which may be the only data rate that allows connectivity. By using the last used data rate irregardless of the transmission result, the first packet would fail and the second packet would reach data rate 0 on the third attempt).

> It also doesn't specify the interaction with ADR (whose recovery system seems aimed at unconfirmed messages).

I am not sure which interaction you mean. My implementation keeps track of the number of packets sent since last downlink received. If this number >= 64, the AdrAckReq bit is set, if this number is >= 64+32, the data rate is lowered, etc.
As confirmed packets normally are acknowledged, this number is reset (as we received a downlink, giving the server the chance to add and ADR command).
If for some reason we have sent 64 confirmed packets that all were not acknowledged, my implementation will still set the AdrAckReq bit. If transmission of confirmed packets is continued, the data rate will be decremented by both the ADR mechanism as well as the confirmed packets mechanism. My implementation prevents that the data rate is decremented twice for one retransmission (by both mechanisms). This behavior is not specified in the specification, if I recall correctly.
If I remember correctly, the data rate used for transmitting the (64+32)th packets will be lowered twice as fast as normal my implementation: If the confirmed packets mechanism (that uses the table specified in chapter 18.4) does not decrement the data rate, the ADR mechanism will (as this is the (64+32)th packet). Instead of decrementing the data rate every two transmission attempts, the data rate is decremented every attempt. This could be a bug :P

Hope this helps :)

brocaar, 2018-06-05:
@tommas-factorylab I have checked today with some people within the LoRa Alliance:

---

An ACK on a confirmed-data message guarantees that the frame has been delivered, however a confirmed-data message does not guarantee that the message will be delivered. If data-consistently is required, then this must be implemented in the application-layer.

With LoRaWAN 1.0 the network-server has the right to see a retransmitted confirmed uplink as a replay attack and thus drop this frame, to keep the network secure. When a device does not deal with this correctly (e.g. ending up in an endless retry loop), then this must be considered as a bug in the device firmware.

LoRaWAN 1.1 has better mechanisms to deal with this, as I already have pointed out in previous comments (adding the channel + DR into account when generating the MIC so that the network-server can distinguish a retransmission from a replay-attack).

---

Brulath, 2018-06-05:
Hey @brocaar,

I understand they've acknowledged the flaw in the standard previously, but I don't get the feeling they understand the implications of the suggestion (if I'm understanding it correctly).

Are they suggesting that vendors and purchasers of any devices which rely on guaranteed delivery of messages need to implement an additional custom protocol on top of LoRaWAN to support loraserver? I'm not aware of any other network server which requires this, so it would be a loraserver-specific variant of firmware and client application (which, typically, would just be a passive data feed, but is now an active part of the session). An additional protocol layer would impact available data space in messages (dwell time limits) and increase RX windows (to allow the remote server to respond), which would impact expected battery life.

Do they suggest disabling frame counters for all existing devices, so they don't back off to data rate 0 eventually (18.4)?

FurqanHKhan, 2018-09-23:
Hi @brocaar !

I'm really confused about the above explanation of ACK transmission by the NS. I will be grateful if you please explain it a bit more. Please check the below figure,

[image]

The figure is taken from the explain scenario of confirmed UL tx in LoRaWAN1.1. As you can when the first ACK is unsuccessful the gateway will resend the ACK frame with an incremented DL frame counter value for the same ConfirmedData0 (Cu) again. However, above you said that gateway will never send the ACK again for the same confirmed UL frame. Also its there in the protocol specification as well that the ACKs are never re-transmitted. But then I want to know that why did included the above example of LoRaWAN operation in the LoRaWANv1.1 specification document?

Waiting for your response on this...

valentinadivincenzo, 2018-11-09:
Hi @brocaar, 
I am also confused about the topic, especially because of the example shown by @FurqanHKhan in the previous post.

Can you give us any clarification?
Thank you!

mdroberts1243, 2020-09-23:
Sorry to reopen this (three years is long in the tooth!), but I'm encountering the same issue currently.  When we run in confirmed mode and we miss an ACK, we re-try 7 more times (8 in total), as-per-the-recommendations, hoping to receive an ACK.  This burns battery/airtime/etc. and isn't really good practice.

In reading this thread I see two things:

1. There is a perceived flaw in the standard, so leave it as is, meaning that the poor end-device is going to kill it's battery with retransmissions every time a single ACK is missed; and
2. Basically, the prevailing view that 1.0.2 devices should never run confirmed. 

So the question is, what is the point of even trying to retransmit at all?  If you will never receive an ACK then you're toast.  I think a solution should favor the (mostly) battery operated devices.  There are lots of other ways malicious actors could deny service so blocking this one vector doesn't seem productive overall.

This could be important for larger deployments of ChirpStack.