brocaar/chirpstack-network-server #273: Downlinks are only transmitted after a second uplink

tommas-factorylab, 2017-11-22:
### Is this a bug or a feature request?
Feature
### What did you expect?
A node I use can request a answer from an application running on a server. This answer has to be transmitted in either RX1 or RX2 as depends on the time. The answer takes up to 100ms to generate and is transmitted to the lora application server using the MQTT "application/+/node/+/tx" topic. I expect the answer to be transmitted in the next RX1 or RX2, depending on the settings of the node. 
### What happened?
The answer is added to the downlink queue and is transmitted only after the next packet is received by the network server. This causes the answer to be outdated and useless. 
### What version are your using?
loraserver version 0.19.0
### How can your issue be reproduced?
Write an application that answers uplink packets with the frame count of the uplink packet. Any downlink will contain the frame count of the uplink of one packet earlier.
### Could you share your log output?
MQTT output:
```
gateway/+/rx {"rxInfo":{"mac":"*","time":"0001-01-01T00:00:00Z","timestamp":3309996588,"frequency":868500000,"channel":2,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-79,"loRaSNR":8.5,"size":32,"dataRate":{"modulation":"LORA","spreadFactor":9,"bandwidth":125}},"phyPayload":"QDKH1AaAAgABPJ8RvZdKYyd1VkIIqP+pQJOZHk5Yw9Q="}
application/17/node/aaaaaaaaaaaaaaaa/rx {"applicationID":"17","applicationName":"*","nodeName":"*","devEUI":"aaaaaaaaaaaaaaaa","rxInfo":[{"mac":"*","rssi":-79,"loRaSNR":8.5,"name":"*","latitude":0,"longitude":0,"altitude":0}],"txInfo":{"frequency":868500000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":9},"adr":true,"codeRate":"4/5"},"fCnt":2,"fPort":1,"data":"*"}
application/17/node/aaaaaaaaaaaaaaaa/tx {"confirmed": false, "data": "*", "fPort": 6, "reference": "aaaaaaaaaaaaaaaa_1511307081.18"}
gateway/+/rx {"rxInfo":{"mac":"*","time":"0001-01-01T00:00:00Z","timestamp":3339995316,"frequency":868100000,"channel":0,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-81,"loRaSNR":12,"size":32,"dataRate":{"modulation":"LORA","spreadFactor":9,"bandwidth":125}},"phyPayload":"QDKH1AaAAwABLlWyCu4vAHFRvcuiuXVfhKPcdIUnHuM="}
application/17/node/aaaaaaaaaaaaaaaa/rx {"applicationID":"17","applicationName":"*","nodeName":"node1","devEUI":"aaaaaaaaaaaaaaaa","rxInfo":[{"mac":"*","rssi":-81,"loRaSNR":12,"name":"*","latitude":0,"longitude":0,"altitude":0}],"txInfo":{"frequency":868100000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":9},"adr":true,"codeRate":"4/5"},"fCnt":3,"fPort":1,"data":"*"}
gateway/+/tx {"txInfo":{"mac":"*","immediately":false,"timestamp":3341995316,"frequency":869525000,"power":14,"dataRate":{"modulation":"LORA","spreadFactor":12,"bandwidth":125},"codeRate":"4/5","iPol":null},"phyPayload":"YDKH1AYAAAAGayXpe6lY3zgeLhc="}
```
The LoRa app-server:
```
Nov 21 21:12:31 * lora-app-server[14421]: time="2017-11-21T21:12:31+01:00" level=info msg="handler/mqtt: publishing data-up payload" topic="application/17/node/aaaaaaaaaaaaaaaa/rx"
Nov 21 21:12:31 * lora-app-server[14421]: time="2017-11-21T21:12:31+01:00" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=aaaaaaaaaaaaaaaa
Nov 21 21:12:31 * lora-app-server[14421]: time="2017-11-21T21:12:31+01:00" level=info msg="handler/mqtt: data-down payload received" topic="application/17/node/aaaaaaaaaaaaaaaa/tx"
Nov 21 21:12:31 * lora-app-server[14421]: time="2017-11-21T21:12:31+01:00" level=info msg="downlink queue item enqueued" dev_eui=aaaaaaaaaaaaaaaa id=34
Nov 21 21:13:01 * lora-app-server[14421]: time="2017-11-21T21:13:01+01:00" level=info msg="handler/mqtt: publishing data-up payload" topic="application/17/node/aaaaaaaaaaaaaaaa/rx"
Nov 21 21:13:01 * lora-app-server[14421]: time="2017-11-21T21:13:01+01:00" level=info msg="downlink queue item deleted" id=34
Nov 21 21:13:01 * lora-app-server[14421]: time="2017-11-21T21:13:01+01:00" level=info msg="data-down item requested by network-server" confirmed=false dev_eui=aaaaaaaaaaaaaaaa fcnt=0 id=34
```

It appears that the network server queries the downlink queue of the application server ASAP. As the application takes up to 100 ms to generate the answer, the answer is transmitted to the application server after the network server issued the downlink queue.
I think that the downlink queue should be queried just before the deadline for RX1/RX2 and not ASAP. This allows any application to generate an answer to be transmitted to the node within the same "cycle". On the other hand, it is important that any transmission commands to the gateway are issued before RX1/RX2 closes. This results in some deadline before which any answer from the application must be transmitted to the application server. The deadline for RX1/RX2 depends on the latency of the gateway, the internal processing time of the gateway, etc. A dynamic algorithm to determine this deadline would be ideal, but for example the following would suffice:
deadlineRx1 = RxDelay / 2
deadlineRx2 = RxDelay / 2 + 1

brocaar, 2017-11-22:
I'm currently in the process of refactoring the downlink queue, also with future class b in mind. This will be one of the things that might get improved. Please note that you can increase the following config value to give the application-server more time to enqueue data as a response to an uplink:

```
--get-downlink-data-delay value         delay between uplink delivery to the app server and getting the downlink data from the app server (if any) (default: 100ms) [$GET_DOWNLINK_DATA_DELAY]
```

https://docs.loraserver.io/loraserver/install/config/

tommas-factorylab, 2017-11-22:
Ah, that configuration parameter fixes my problem! Thank you!

closed 2017-11-22 by brocaar

sputnik-alarm, 2017-12-08:
Is there any rules, of how the confirmed downlink with ack getting timeout will be retransmitted again?
Does server generally makes this retransmition? How much attempts will make server? Is there any dependencies on working in different LoRa classes?

brocaar, 2017-12-08:
I've just released a new version which sends either an ACK or nACK. Then it is up to your application if you want to re-transmit or not. The server won't schedule any re-transmissions anymore.