brocaar/chirpstack-network-server #274: No variable to set AS host:port

bogdanoniga, 2017-11-22:
### What did you expect?
A variable to set the AS host:port.
https://github.com/brocaar/loraserver/blob/master/cmd/loraserver/main.go#L595-L609

### What happened?
I was running the servers using Docker and when publishing a message to Loraserver, it is trying to connect to Appserver on 127.0.0.1:8001 (by default) but my Appserver is in another container and there is no EnvVar to set the AS host:port like NS_SERVER EnvVar for setting the host:port for NS.

### What version are your using?
Latest version.

### Could you share your log output?
```
loraserver_1       | time="2017-11-22T21:44:09Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {localhost:8001 <nil>}"
```

iegomez, 2017-11-22:
That's defined in the lora-app-server configuration and stored at the loraserver DB in the routing_profile table. The specific flags are --as-public-server for the host:ip part, and --as-public-id for an AS id. You can check the docs on this here: https://docs.loraserver.io/lora-app-server/install/config/

bogdanoniga, 2017-11-22:
@iegomez it worked, thank you

closed 2017-11-22 by bogdanoniga