brocaar/chirpstack-network-server #276: No ACK for valid packet

alexanderplanar, 2017-12-06:
### Is this a bug or a feature request?
Bug.

### What did you expect?
Receive an acknowledgement.

### What happened?
The server does not send ACK packet actually, but registers the reply packet in the "frames log".

### What version are your using?
loraserver version 0.22.0
lora-app-server version 0.14.1

### How can your issue be reproduced?
I do not know. It is difficult to reproduce.

### Could you share your log output?
Yes. See attached files.
[framesCSV_full_logs.zip](https://github.com/brocaar/loraserver/files/1535259/framesCSV_full_logs.zip)
[postgre_DBs.zip](https://github.com/brocaar/loraserver/files/1535260/postgre_DBs.zip)
[server_log_full.zip](https://github.com/brocaar/loraserver/files/1535261/server_log_full.zip)
[ShortLogs.zip](https://github.com/brocaar/loraserver/files/1535262/ShortLogs.zip)
[tcpdump_full_logs.zip](https://github.com/brocaar/loraserver/files/1535263/tcpdump_full_logs.zip)

alexanderplanar, 2017-12-06:
Hello!

Could you help me to understand the problem.
I use a confirmed UpLink messages.
loraserver version is 0.22.0
lora-app-server version is 0.14.1
The programs are running on Debian OS inside Virtual Box Machine.
The Gateway based on BeagleBone board connected to Microchip Radio Board SX1301.
The Server and the Gateway are in the same network, VM uses bridged network adapter.
The hosts use NTP to time syncronisation.
I noticed that sometimes some packets are not confirmed.
The Node trys to retransmit the packet 8 times, but with no luck.
I catched the following logs: tcpdump, syslog from network/app server and frame logs.
I pass the TCPDUMP logs trough my script to filter out all LoRa packets.
If it required I can upload this script, but it is just "tcpdump -i eth2 -v -XX --immediate-mode <dir> host <myhost>"
 1) TCPDUMP log on gateway host:
      LoRaLog-GatewayMachine-UpStream
      LoRaLog-GatewayMachine-DownStream
      I pass the packets trough my script to filter out all LoRa packets.
 2) TCPDUMP log on server host:
      LoRaLog-ServerMachine-UpStream
      LoRaLog-ServerMachine-DownStream
      I pass the packets trough my script to filter out all LoRa packets.
 3) Server Logs: I just used "sudo tail -f /var/log/syslog | grep lora" to filter out all messages in SysLog corresponing to LoRaServer and ApplicationServer.
 4) FrameLogs from PostgreSQL DB using pgAdmin 3
      I connect to DB (loraserver_ns) and look for frame_log table:
      SELECT id, rx_info_set, tx_info, encode(phy_payload,'hex') FROM public.frame_log WHERE id
      BETWEEN 175000 AND 189000 ORDER BY created_at ASC
      This SQL request gives me the all packets for my case: Dec 6 2017, 9:30 AM and so on.

I found out, that the server "thinks" that it sent an ACK packet, but actually the packet is sent after about 30 second delay... (see tcpdump logs).
If the issue is about MIC or radio path, the packet would be courrupted during retransmissions different way, but all 8 packets are the same. I think, the server rejects the rest packets by FCnt condition.

Could you help to solve the problem?

alexanderplanar, 2017-12-06:
UPD. Do you have any special simple utility to decode payload from LoRa packet using NWKEY and APPKEY?
It would help me to debug the packet exchange.

alexanderplanar, 2017-12-07:
**UPD.** Excuse me, I found a mistake in my script, and it can affect to results. In particular, the timestamp logged in tcpdump logs is shifted relatively to packet. Now I've fixed it and going to start a new test. 
p.s. But it does not mean that the effect I described will disappear. It appears by itself time to time without any tests, I just try to log it and confirm it is real.
As I get any results to upload, I delete old logs from issue.

Thanks!

alexanderplanar, 2017-12-07:
UPD. I found out that the loraserver does not responsible for this bug, at least, It actually send response, but it may be dropped by lora-gateway. The issue can be closed. Thanks.

alexanderplanar, 2017-12-12:
I reopen the issue, because it seems I found an evidence of the loraserver fault.
The packet is processed too long and the reply is transmitted to late to be emitted by gateway.
The gateway rejects the packet by wrong timestamp condition, because it thinks the RX1 is already closed.

The gateway error:

```
JSON down: {"txpk":{"imme":false,"tmst":256928027,"freq":868.3,"rfch":0,"powe":14,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":12,"data":"YPwKEgegOQDEMNh9"}}
src/jitqueue.c:251:jit_enqueue(): ERROR: Packet REJECTED, timestamp seems wrong, too much in advance (current=258209393, packet=256928027, type=0)
ERROR: Packet REJECTED (jit error=2)
lgw_receive:1155: FIFO content: 1 83 2 5 1d
```

frame_log (SQL) log:
```
"id","created_at","rx_info_set","tx_info","encode"
"269677","2017-12-12 14:21:52.66489+05","[{""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""loRaSNR"": 8.8, ""timestamp"": 255928027, ""crcStatus"": 1, ""mac"": ""aa555a0000000101"", ""frequency"": 868300000, ""rfChain"": 1, ""codeRate"": ""4/5"", ""time"": ""0001-01-01T00:00:00Z"", ""rssi"": -91, ""channel"": 1, ""size"": 29}]","null","80fc0a12078039000209e40e7e6e15d6447458d8a4c733143d0d98c4da"
"269678","2017-12-12 14:21:53.898056+05","null","{""iPol"": null, ""power"": 14, ""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""timestamp"": 256928027, ""mac"": ""aa555a0000000101"", ""frequency"": 868300000, ""codeRate"": ""4/5"", ""immediately"": false}","60fc0a1207a03900c430d87d"
"269679","2017-12-12 14:22:31.694332+05","[{""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 10}, ""loRaSNR"": 10, ""timestamp"": 294968388, ""crcStatus"": 1, ""mac"": ""aa555a0000000101"", ""frequency"": 868100000, ""rfChain"": 1, ""codeRate"": ""4/5"", ""time"": ""0001-01-01T00:00:00Z"", ""rssi"": -95, ""channel"": 0, ""size"": 29}]","null","80fc0a1207803a00028069d57440a1aaceef359fb78853c76a9a9838d8"
"269680","2017-12-12 14:22:32.017671+05","null","{""iPol"": null, ""power"": 14, ""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 10}, ""timestamp"": 295968388, ""mac"": ""aa555a0000000101"", ""frequency"": 868100000, ""codeRate"": ""4/5"", ""immediately"": false}","60fc0a1207a53a0003520700013a92c7b3"
"269681","2017-12-12 14:22:34.1969+05","[{""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""loRaSNR"": 7.5, ""timestamp"": 297653267, ""crcStatus"": 1, ""mac"": ""aa555a0000000101"", ""frequency"": 868500000, ""rfChain"": 1, ""codeRate"": ""4/5"", ""time"": ""0001-01-01T00:00:00Z"", ""rssi"": -85, ""channel"": 2, ""size"": 31}]","null","80fc0a1207823b0003070283d7e610e1c0fa610104cd14e52ace8bb90787a9"
"269682","2017-12-12 14:22:34.460872+05","null","{""iPol"": null, ""power"": 14, ""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""timestamp"": 298653267, ""mac"": ""aa555a0000000101"", ""frequency"": 868500000, ""codeRate"": ""4/5"", ""immediately"": false}","60fc0a1207a53b000354070001bec23c55"
"269683","2017-12-12 14:22:36.089553+05","[{""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""loRaSNR"": 7.8, ""timestamp"": 299110059, ""crcStatus"": 1, ""mac"": ""aa555a0000000101"", ""frequency"": 868500000, ""rfChain"": 1, ""codeRate"": ""4/5"", ""time"": ""0001-01-01T00:00:00Z"", ""rssi"": -89, ""channel"": 2, ""size"": 31}]","null","80fc0a1207823c00030702985a869e259db5d69dbb04395f7cbc685bfd22c3"
"269684","2017-12-12 14:22:36.772912+05","null","{""iPol"": null, ""power"": 14, ""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""timestamp"": 300110059, ""mac"": ""aa555a0000000101"", ""frequency"": 868500000, ""codeRate"": ""4/5"", ""immediately"": false}","60fc0a1207a53c0003560700018d687224"
"269685","2017-12-12 14:23:10.663303+05","[{""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 10}, ""loRaSNR"": 13.5, ""timestamp"": 334177836, ""crcStatus"": 1, ""mac"": ""aa555a0000000101"", ""frequency"": 868100000, ""rfChain"": 1, ""codeRate"": ""4/5"", ""time"": ""0001-01-01T00:00:00Z"", ""rssi"": -89, ""channel"": 0, ""size"": 29}]","null","80fc0a1207803d0002b681e8dce9b4946d22388db97e26229fadca6989"
"269686","2017-12-12 14:23:10.843752+05","null","{""iPol"": null, ""power"": 14, ""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 10}, ""timestamp"": 335177836, ""mac"": ""aa555a0000000101"", ""frequency"": 868100000, ""codeRate"": ""4/5"", ""immediately"": false}","60fc0a1207a53d00035307000107ea12fe"
"269687","2017-12-12 14:23:13.207872+05","[{""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""loRaSNR"": 7.2, ""timestamp"": 336720739, ""crcStatus"": 1, ""mac"": ""aa555a0000000101"", ""frequency"": 868500000, ""rfChain"": 1, ""codeRate"": ""4/5"", ""time"": ""0001-01-01T00:00:00Z"", ""rssi"": -87, ""channel"": 2, ""size"": 31}]","null","80fc0a1207823e000307026b014fe2b0033341169af83a352fd40c36495e32"
"269688","2017-12-12 14:23:13.395829+05","null","{""iPol"": null, ""power"": 14, ""dataRate"": {""modulation"": ""LORA"", ""bandwidth"": 125, ""spreadFactor"": 7}, ""timestamp"": 337720739, ""mac"": ""aa555a0000000101"", ""frequency"": 868500000, ""codeRate"": ""4/5"", ""immediately"": false}","60fc0a1207a53e000356070001f7010c21"
[... cut ...]
### Rx Packet Timeout
ReSend on ACK timeout (7 < 8)
TuneFreq: 868500 kHz
### Rx Packet Timeout
Give up on ACK timeout (8 >= 8)
TuneFreq: 868100 kHz
ACK received (1)
UpLink counter: 62
```

More info is in attach files.
[attach.zip](https://github.com/brocaar/loraserver/files/1551703/attach.zip)

Do you have any ideas, why the server process the packet so long time?
I understand, that the server is running under VM, but the usual process time is about 0,3 sec, this packet was being processed during 1.3 sec.

brocaar, 2017-12-12:
Could you post your loraserver settings? E.g. a long de-duplication delay could be the cause of this (as you instruct LoRa Server to wait for a long time). Also the downlink data delay adds to the time to the roundtrip. An other thing to consider, what is the roundtrip time between your gateway and the network-server? In some cases when you're using a 3G connection, it makes sense to configure the `--rx1-delay` to something > 1 second.

alexanderplanar, 2017-12-13:
Hi, @brocaar, thanks for your fast reply!

Yes, sure :)

I use local network between LoRa Server and Gateway, no VPN or 3G is used, Ethernet is used only.
But the LoRa Server is running under VirtualBox with using bridged adapter. Maybe, it contributes some latency.

I do not override the default value of key _--rx1-delay_. it is by default - 1 second.

I updated applications recently:
lora-app-server version 0.14.1
loraserver version 0.22.0

So, I will try to move my server to a host with better performance.

my **/etc/default/loraserver** file:
```
# network identifier (NetID, 3 bytes) encoded as HEX (e.g. 010203)
NET_ID=010203

# ism band configuration to use (options: AU_915_928, CN_470_510, EU_863_870, US_902_928)
BAND=EU_863_870

# ca certificate used by the api server (optional)
CA_CERT=

# tls certificate used by the api server (optional)
TLS_CERT=

# tls key used by the api server (optional)
TLS_KEY=

# ip:port to bind the api server (default: "0.0.0.0:8000")
BIND=0.0.0.0:8000

# redis url (e.g. redis://user:password@hostname:port/0) (default: "redis://localhost:6379")
REDIS_URL=redis://localhost:6379

# postgresql dsn (e.g.: postgres://user:password@hostname/database?sslmode=disable)
POSTGRES_DSN=postgres://loraserver_ns:***@localhost/loraserver_ns?sslmode=disable

# automatically apply database migrations
DB_AUTOMIGRATE=true

# mqtt broker server used by the gateway backend (e.g. scheme://host:port where scheme is tcp, ssl or ws) (default: "tcp://localhost:1883")
GW_MQTT_SERVER=tcp://localhost:1883

# mqtt username used by the gateway backend (optional)
GW_MQTT_USERNAME=sammy

# mqtt password used by the gateway backend (optional)
GW_MQTT_PASSWORD=***

# hostname:port of the application-server api server (optional) (default: "127.0.0.1:8001")
AS_SERVER=127.0.0.1:8001

# ca certificate used by the application-server client (optional)
AS_CA_CERT=

# tls certificate used by the application-server client (optional)
AS_TLS_CERT=

# tls key used by the application-server client (optional)
AS_TLS_KEY=

# hostname:port of the network-controller api server (optional)
NC_SERVER=

# ca certificate used by the network-controller client (optional)
NC_CA_CERT=

# tls certificate used by the network-controller client (optional)
NC_TLS_CERT=

# tls key used by the network-controller client (optional)
NC_TLS_KEY=

# time to wait for uplink de-duplication (default: 200ms)
DEDUPLICATION_DELAY=200ms

# delay between uplink delivery to the app server and getting the downlink data from the app server (if any) (default: 100ms)
GET_DOWNLINK_DATA_DELAY=100ms

# timezone to use when aggregating data (e.g. 'Europe/Amsterdam') (optional, by default the local timezone is used)
# TIMEZONE=Europe/Amsterdam

# create non-existing gateways on receiving of stats
GW_CREATE_ON_STATS=true

# aggregation intervals to use for aggregating the gateway stats (valid options: second, minute, hour, day, week, month, quarter, year)
GW_STATS_AGGREGATION_INTERVALS=minute,hour,day

# extra frequencies to use for ISM bands that implement the CFList
# EXTRA_FREQUENCIES=867100000,867300000,867500000,867700000

# enable only a given sub-set of channels (e.g. '0-7,64')
# ENABLE_UPLINK_CHANNELS=0-7,64

# the ttl after which a node-session expires after no activity (default 31 days)
NODE_SESSION_TTL=744h0m0s

# log uplink and downlink frames to the database
LOG_NODE_FRAMES=true

# ca certificate used by the gateway api server (optional)
GW_SERVER_CA_CERT=

# tls certificate used by the gateway api server (optional)
GW_SERVER_TLS_CERT=

# tls key used by the gateway api server (optional)
GW_SERVER_TLS_KEY=

# JWT secret used by the gateway api server for gateway authentication / authorization
# You could generate this by executing 'openssl rand -base64 32' for example
GW_SERVER_JWT_SECRET=***

# ip:port to bind the gateway api server (default: "0.0.0.0:8002")
GW_SERVER_BIND=0.0.0.0:8002

```

alexanderplanar, 2017-12-13:
**UPD.** I moved the VM to the host with the higher performance, no luck. 4 lost ACKs per 5 minutes (about 200 messages).

alexanderplanar, 2017-12-14:
I tried to create a full log of error, but came across to the strange problem.
I noticed that the timestamps (tmst) on gateway log does not relate to the real delay of roundtrip.
I suggested that the full roundtrip between the gateway and the server equals the difference btw a couple values of tmst on gateway, but it is not. The gateways reports that the packet has been rejected due to the timestamps "is too mush in advance", but the difference between timestamps is different each time: 120ms, 200ms, 1.2sec.... Why gateway rejects the packet in case the difference between timestamps is 120ms? In the same time the server reports that only the processing time is about 990 msec (frame_logs)...
[attach.zip](https://github.com/brocaar/loraserver/files/1559411/attach.zip)

Maybe, it is seems that it is required to adjust the RXDELAY1 and RXDELAY2 to get the network works stable... 

I will try to update the apps to the new version (15) tomorrow

alexanderplanar, 2017-12-15:
Hello, @brocaar !
Thanks for advice :-)
Could you explain, if the LoRa Gateway Bridge is the proxy-like application, which will be retranslate the packets between the LoRa Network Server and LoRa Gateway?
Now on my Beaglebone the only application is running: LoRa Gateway Packet Forwarder. It conects to the LoRa Network Server directly using UDP Semtech. Do I understand correctly, LoRa Gateway Bridge will wedge into the exchange between the gateway and server?

alexanderplanar, 2017-12-15:
@brocaar, I've updgraded the apps, but with no results. The problem is still actual.
Current used versions: 
loraserver version 0.23.0
lora-app-server version 0.15.0

By the way, I want to note, I changed the application duty cycle, so my Node sends about one packet per second, of course in case the ACK is received by the Node. It causes the problem to show up fast.

So, I also will try to configure LoRa Gateway Bridge to get more information.

alexanderplanar, 2017-12-20:
Hello,
I did some tests with LoRa Gateway Bridge on the gateway host, but with no luck.
The problem is still actual. In particular, I see ACK, TX and RX events when I type 'mosquitto_sub -v -t "gateway/+/#", but I do not see nACKS. In the same time the nACKS actually are, because I see JIT-errors in the packet forwarder logs, and I see the packet retransmissions.

brocaar, 2017-12-21:
> But the problem with the packet retransmissions is still actual.

Just to make sure I don't make any assumptions, you mean that LoRa Server does not ACK a confirmed data re-transmission?

alexanderplanar, 2017-12-21:
Hello, @brocaar!
No, I mean, the server does respond to confirmed uplink, but too late to the ACK packet can be transmitted by the gateway. So, the gateway rejects the packet by the timestamp condition (JIT error = 2, timestamp seems wrong). Therefore the server "thinks" that the packet already confirmed (the ACK has been emitted by server), but actually the gateway rejected the packet. And the node unable to determine if the packet lost on uplink path, or downlink path, and begins the long procedure of retransmission. But each retransmission must be ignored by server, because server "thinks" the packet already confirmed and does not respond.

alexanderplanar, 2017-12-21:
have to note, this situation happens time to time, and the system works in general. But sometimes something goes wrong and the server process the packet longer than usually.

brocaar, 2017-12-21:
Yes, I'm aware of that and I will improve this. For the time being, you could increase the `--rx1-delay` setting so you have more time for the roundtrip. When you know that you never have to wait for the application-server to enqueue data, directly after an uplink, you could also set `--get-downlink-data-delay` to `0`.

I'm also planning to add more debug logging with timing, so it is easier to debug which function consumes what amount of time.

For the gateway communication, the first effort was to expose the `gateway/MAC/ack` topic, so this feedback can be used by LoRa Server in the future :-)

alexanderplanar, 2017-12-21:
Ok, thanks for advice, I will try to change RXDELAY1. But if I change rxdelay1, would I need to change rxdelay2? As I understand correct, the RXDELAY1 and RXDELAY2 is related to the same moment of time: the time of transmitting, RXDELAY2 = RXDELAY1 + 1, isn't it?

brocaar, 2017-12-21:
That is correct and you only have to set the rx1 delay, as the rx2 delay is always rx1 delay + 1 second.

alexanderplanar, 2017-12-21:
Perfect, thanks! ;-)

alexanderplanar, 2018-01-09:
Happy New Year everyone!

@brocaar, Hello!
I made a test at holydays. I changed the RXDELAY1 to 2 seconds. Here is the result:
Duration: 		    11 days (949064 seconds)
Packets sent:   	    300655
Packets without ACK:  953 (0.3170%)
RX Timeouts:     	    7942
Packets received:         300654
RX Errors:                     692

I think is not bad :) but is not ideally. 
I double checked the last lost packet and got see that the packet processed fast enough (about 200ms -  from 'frame_logs' table in DB). Then I checked the packet forwarder log, and did not found any records telling the packet had been rejected.  I thought the issue is the ACK-packets are rejected by packet_forwarder by JIT error and so it is not emitted. But now I use a new version of network server (0.23), and I do not see such errors . So the ACK has been sent apparently. But the node does not receive it due to some reason. Maybe it is the firmware issue, maybe, it depends on some kind of ether noise, I do not know for sure. More over I see lots of errorneous RX-packets.
You said a couple weeks ago that you going to improve something in network server relating to the packet processing. 
Maybe it is better to do a new long test after this fix to be sure that the problem does not refers to network server at all. Could you tell when the new version will be released approximately? 

Thanks a lot!

brocaar, 2018-01-22:
> You said a couple weeks ago that you going to improve something in network server relating to the packet processing.
Maybe it is better to do a new long test after this fix to be sure that the problem does not refers to network server at all. Could you tell when the new version will be released approximately?

Yes, I want to handle downlink messages in a smarter way, but this should only affect downlink transmissions that initially failed because of a nACK by LoRa Gateway Bridge / the packet-forwarder. E.g. in case one gateway is unable to transmit, re-try using RX2 parameters (if possible) or a different gateway on RX1 or RX2. I don't have an ETA for this yet.

brocaar, 2018-05-10:
Is this issue still present? Else I will close this issue.

closed 2018-10-16 by brocaar