brocaar/chirpstack-network-server #277: Rules of confirmed downlinks retransmition if no ACK from node

sputnik-alarm, 2017-12-08:
### Is this a bug or a feature request?
Feature.

### What did you expect?
Whant server to retransmit confirmed downlinks if node not answering.

### What happened?
Server doesn't retransmit confirmed downlinks.

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

sputnik-alarm, 2017-12-08:
Is there any rules, of how the confirmed downlink with ack getting timeout will be retransmitted again?
Does server generally makes this retransmition? How much attempts will make server? Is there any dependencies on working in different LoRa classes?

brocaar, 2017-12-08:
When using the latest LoRa Server version, it will never re-transmit a confirmed downlink. Also LoRa Server would not be able to do so, as the payload is encrypted using the downlink frame-counter, which can not be re-used.  Please see: https://forum.loraserver.io/t/release-lora-app-server-0-15-lora-server-0-25/492.

closed 2017-12-08 by brocaar