brocaar/chirpstack-network-server #278: LoRaserver 0.23.0 bug

Richaagarwalrn, 2017-12-12:
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.23.0
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="connecting to postgresql"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="setup application-server client pool" ca_cert= tls_cert= tls_key=
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="configuring join-server" ca_cert= server="http://localhost:8003" tls_cert= tls_key=
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="no network-controller configured"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="applying database migrations"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="backend/gateway: connected to mqtt server"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="backend/gateway: subscribing to stats topic" topic="gateway/+/stats"
Dec 12 10:41:50 ubuntu loraserver[28654]: time="2017-12-12T10:41:50+05:30" level=info msg="migrations applied" count=0
Dec 12 10:41:50 ubuntu systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Dec 12 10:41:50 ubuntu systemd[1]: loraserver.service: Unit entered failed state.
Dec 12 10:41:50 ubuntu systemd[1]: loraserver.service: Failed with result 'exit-code'.
Dec 12 10:41:51 ubuntu systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Dec 12 10:41:51 ubuntu systemd[1]: Stopped LoRa Server.
Dec 12 10:41:51 ubuntu systemd[1]: loraserver.service: Start request repeated too quickly.
Dec 12 10:41:51 ubuntu systemd[1]: Failed to start LoRa Server.

Richaagarwalrn, 2017-12-12:
Loraserver repeatedly stop in newer release. Please suggest how to proceed

brocaar, 2017-12-12:
I've already replied to https://forum.loraserver.io/t/release-lora-app-server-0-15-lora-server-0-23/492.

closed 2017-12-12 by brocaar