brocaar/chirpstack-network-server #279: Multicast Downlink to XDOT devices

habbasi3, 2017-12-13:
Hi,

Thank you for this great work. I would like to send a multicast downlink message to multiple XDots through a single Multitech gateway. From your previous post, I understand that I should setup a multicast node using LoRa AppServer (using ABP and Class C) and configure all of my nodes with this multicast configuration (DevAddr, NwkSKey, AppSKey, etc...) to send a multicast downlink. However, XDots have factory defined DeviceEUI (which cannot be changed). So, how can I setup multiple XDOTs to have the same DevAddr or send a multicast message to multiple XDots with different DeviceEUI ?

Thanks a lot

brocaar, 2017-12-13:
Mind to re-post this question in the forum? https://forum.loraserver.io. Other people might benefit from the answer too :-) Github issues are fine for bugs, but after closing an issue the answer becomes "invisible" making it less ideal for a knowledgebase.

closed 2017-12-13 by brocaar

habbasi3, 2017-12-13:
Hi,

I have posted the question in the loraserver forum. Please check.

Thanks a lot