brocaar/chirpstack-network-server #292: Build source

Almaz1c, 2018-02-06:
Hello!
Trying to build from sources.
Go and Protobuf are installed successfully.
`make requirements` also passed without errors.
There are errors during `make test`:

> --- PASS: TestADR (0.03s)
> PASS
> ok      github.com/brocaar/loraserver/internal/adr      0.041s
> === RUN   TestGatewayAPI
> 
>   Given a clean database and api instance ERRO[0000] ping database error, will retry in 2s: pq: password authentication failed for user "root"
> ERRO[0002] ping database error, will retry in 2s: pq: password authentication failed for user "root"
> 

What is a next steps for installation? Calling `packaging/package.sh` returns me `Unknown target`
as `GOARCH="amd64"`

brocaar, 2018-02-06:
Please see the provided `docker-compose.yml` file for setting up a test environment, including test databases. The commands are explained here (including for packaging): https://docs.loraserver.io/loraserver/community/source/.

closed 2018-02-06 by brocaar

Almaz1c, 2018-02-06:
Wow! Adding `export TEST_POSTGRES_DSN="postgres://loraserver_ns:your_pass@localhost/loraserver_ns?sslmode=disable"
`
variable to bash lets `make test` to pass all tests! Awesome! Thank you very much for awesome software!