brocaar/chirpstack-network-server #294: implementation of TDOA and RSSI based time domain geolocation

narioinc, 2018-02-14:
### Is this a bug or a feature request?
This is a feature request.

### What did you expect?
Is there a plan for the Lora server to implement the new TDOA and RSSI geolocation feature announced by Semtech during the TTN conference ?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

gillespilloudwyres, 2018-04-05:
Hi,

As I have understood, the new precise-timestamp sent by semtech chip will be under licence, and an decrypt key, will be necessary to have the precise timestamp requested to do TDOA. Have you hear that ?

narioinc, 2018-04-07:
Yes, I haven't yet looked at the complete business model being advertised by semtech for their blob license. But I guess we will soon find out once we get boards that support the feature. I'll keep this issue updated when I get additional information.

narioinc, 2018-04-07:
TTN is yet to get a full fledged implementation out...so I am still waiting for OEM to support the feature.

brocaar, 2018-09-12:
Support for an external geolocation server has been merged into the `master` branch. I'll announce a geolocation server (which integrates with Collos) soon :)

closed 2018-09-12 by brocaar

brocaar, 2018-09-17:
Please see https://forum.loraserver.io/t/release-lora-app-server-v2-2-0-lora-geo-server-v2-0-0/2246.

commit 0a32d460 (2018-09-12): Implement support for geolocation server.

Closes #294.