brocaar/chirpstack-network-server #297: Version checking for PostgreSQL, Redis and other components

glederer007, 2018-02-22:
Hello Orne,

This is a feature request.

I often notice in the forum that users are installing loraserver / lora-app-server with the incorrect version of psql or other components. Would it possible to include some kind of component version checking to both loraserver and lora-app-server? I mean that loraserver / lora-app-server could check at start for the version of psql / redis / mqtt broker / etc. and if the version is not correct, a big welcome message would tell them the problem. I think this could save a lot of efforts on your side as well.

Regards,

brocaar, 2018-03-09:
Yes, such a thing could be nice :-)

brocaar, 2019-02-13:
I'm closing this issue. Most distros ship with an up-to-date Redis and PostgreSQL package and this hasn't been popped up for a long time on the forum.

closed 2019-02-13 by brocaar