brocaar/chirpstack-network-server #300: ABP Node Activation Values get cleared from server

shivams, 2018-03-01:
This is a Bug report. 

In the "Node Activation" tab of an ABP node, the "device address", "network session key" and "application session key" values disappear sometimes for some nodes. Why does this happen?

brocaar, 2018-03-02:
This can be anything (depending your Redis configuration). But also look at: https://docs.loraserver.io/loraserver/install/config/ `device_session_ttl`

closed 2018-03-02 by brocaar

vincems, 2019-03-21:
Seeking clarification

Hi - I'm having a similar situation.

We have many devices batch loaded and activated via the API but sitting powered off.
If they are not deployed within the TTL time are you saying we have to re-activate them ?   

That in itself is not a problem but it seems a bit odd that considering the database already contains the keys for each device - why should we have to re-add them ?

Also the device ID disappears too - that should not time out and disappear I would have thought.

All the data to reactivate it is actually still in the DB but I've not figured out how to re-activate using the existing values.

Thoughts ?

Cavia68, 2022-09-18:
> This can be anything (depending your Redis configuration). But also look at: https://docs.loraserver.io/loraserver/install/config/ `device_session_ttl`

@brocaar ---> we are facing the same problem and we cannot found any answer to that. (Activation Values get cleared from server after period). Any idea how to solve it?

Cavia68, 2022-09-18:
> Seeking clarification
>
> Hi - I'm having a similar situation.
>
> We have many devices batch loaded and activated via the API but sitting powered off. If they are not deployed within the TTL time are you [...]

Did you find a solution to it?