brocaar/chirpstack-network-server #303: LoRaWAN 1.1 Support

jobrud, 2018-03-05:
Hi,
there are any plans for LoRaWAN 1.1 support ?

Thanks

brocaar, 2018-03-06:
Yes, there are plans to support LoRaWAN 1.1. I've already made some changes to the code-base to make the future implementation of LoRaWAN 1.1 easier, however I'm also dependent on some sort of device support for LoRaWAN 1.1 to test with. As far as I know, https://github.com/Lora-net/LoRaMac-node does not support LoRaWAN 1.1 yet?

4refr0nt, 2018-03-09:
Aslo, see LoRaWAN Regional Parameters v1.1rB (RU region)

jobrud, 2018-03-09:
Thanks for the information. We are currently working on the 1.1 stack. It's planed to publish it in Q3 see the [roadmap](https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Roadmap).

brocaar, 2018-07-03:
A first version for testing has been released. Please see: https://forum.loraserver.io/t/help-testing-the-next-lora-app-server-release-with-lorawan-1-1-support/1610.

closed 2018-07-03 by brocaar