brocaar/chirpstack-network-server #306: DevEUI Uniqueness

1995parham, 2018-03-13:
### Is this a bug or a feature request?
DevEUI must be unique per application by considering AppEUI instead system-wide uniqueness.

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2018-03-13:
This will not work. The AppEUI will be renamed to JoinEUI (already happened in LoRaWAN 1.1) and will be used to identify the entity which handles the OTAA join-request. Therefore the DevEUI must be globally unique.

closed 2018-03-13 by brocaar

1995parham, 2018-03-13:
Thanks for your answer so you think there is no way for providing a way for joining two things with same DevEUI in loraserver.io?

brocaar, 2018-03-13:
No, that won't be supported as it would make a lot of things more complex than needed. Luckily the 64 bit address space should give you enough DevEUIs I hope 😉