brocaar/chirpstack-network-server #307: processing rx packet error: get device-session error: device-session does not exist or invalid fcnt or mic

EtienneBourdeau, 2018-03-15:
Hello everyone !

I am currently using LoRa Server with a Kerlink Wirnet Station (using Semtech Packet Forwarder on the gateway and LoRa Gateway Bridge on the same server than LoRa Server).

I am using OTAA, and configured my gateway and node accordingly, but still, I am seeing this error :

mars 15 12:36:10 debian loraserver[832]: time="2018-03-15T12:36:10+01:00" level=error msg="processing rx packet error: get device-session error: device-session does not exist or invalid fcnt or mic" data_base64="gAAAAAAAAAAIjjSHYGm5vF5/XgdZrJ3/Vioc6Nx5CFY9yhkObRYHPANmWEbCRIlRBIT7YwtaH6IZ/K+2VZ/P8nhsg1Y2jN2ClF2QnsfLbmD10cykd4tv5oFA9dRptl3gMd9LoCBOVLnX2s5nnQb5vjZ8TSXKV3hZR1dyUh42VNiV5FEFnfIszoHrM2I+l9KNq/2vRaQ="

Though, I don't think I can see fcnt on my node, or maybe I don't know how.
Could someone help me ?

Thank you very much !
Regards,
Etienne

arihantdaga, 2018-03-22:
Facing the same issue only sometimes.. randomly..

brocaar, 2018-03-30:
Please see https://forum.loraserver.io/search?q=device-session%20does%20not%20exist%20or%20invalid%20fcnt%20or%20mic.

closed 2018-03-30 by brocaar