brocaar/chirpstack-network-server #313: LoRa Server leaking redis sockets

sa-wilson, 2018-04-05:
Using version 0.25.1 on an x86_64 Ubuntu Server.

LoRa Server is leaking file descriptors for sockets connecting to redis at a rate of approximately 100-200/hour on our system.

This appears to be made worse by viewing the Live Frame Logs (For either gateways or devices) with LoRa App Server, as I can watch the number of open fds increment as I switch on and off the live logs tab.

The following command line output shows that the service has been running since midday, and in almost 4 hours has accumulated 727 open file descriptors.

```
swilson@loraserver:~$ systemctl status loraserver-au915
● loraserver-au915.service - LoRa Server (AU915)
   Loaded: loaded (/etc/systemd/system/loraserver-au915.service; enabled; vendor preset: e
   Active: active (running) since Thu 2018-04-05 11:50:41 AEST; 3h 56min ago
     Docs: https://docs.loraserver.io
 Main PID: 1927 (loraserver)
    Tasks: 15
   Memory: 22.7M
      CPU: 15.496s
   CGroup: /system.slice/loraserver-au915.service
           └─1927 /usr/bin/loraserver -c /etc/loraserver/au915.toml
swilson@loraserver:~$ date && sudo lsof -i -a -p 1927 | wc -l && sudo lsof -i -a -p 1927 | tail -n 1
Thursday 5 April  16:03:39 AEST 2018
727
loraserve 1927 loraserver  729u  IPv4  73013      0t0  TCP localhost:40698->localhost:6379 (ESTABLISHED)
```

The network server in question is running on the AU915 Frequency Plan, using sub-band 3 (Channels 16-23 and 67). We also have an AS923 server running on the same system, although that does not appear to exhibit the same behaviour and only has 7 open fds.

Configurations for the two servers only differ in Postgres user/pass/database, redis username, net id, frequency plan, repeater_compatible, uplink channels, and the bind ports.

The main thing in the loraserver logs is gRPC calls for StreamFrameLogsForGateway
```
Apr 05 15:18:29 loraserver.local loraserver[1927]: time="2018-04-05T15:18:29+10:00" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamFrameLogsForGateway grpc.service=ns.NetworkServer grpc.start_time="2018-04-05T15:18:14+10:00" grpc.time_ms=15101.13 peer.address="[::1]:37780" span.kind=server system=grpc
```

I only have a passing familiarity with Go, but I did take a look and notice that `getFrameLogs()` in `loraserver/internal/framelog/framelog.go` calls `Close()` on the redis socket twice, once deferred, and once more in a goroutine. Could there be an issue caused by this?

brocaar, 2018-04-05:
Thanks for reporting this 👍  I'm able to reproduce the issue and will look into a fix.

sa-wilson, 2018-04-06:
Thanks for finding it so quickly. Good to know that it's not something weird on our end. :+1: 

That makes sense, I'd probably make the same assumption. We'll just have to be careful with the frame log viewer until you get the updated release out.

brocaar, 2018-04-06:
I've also reported this as a (possible) bug in the Redis client library: https://github.com/gomodule/redigo/issues/332

brocaar, 2018-04-06:
It turned out to be an implementation mistake at my side. Calling `.Close()` and `.Receive()` concurrently is not supported by the Redis library.

commit 921bafde (2018-04-06): Fix leaking Redis connections on pubsub.

Closes #313.
closed 2018-04-06 by brocaar