brocaar/chirpstack-network-server #314: RX2 implementation

gillespilloudwyres, 2018-04-05:
Hi !
It could be nice to have the transmission retry on the RX2 window feature.

As defined in 3.3.2 of the lorawan 1.1 spec, the implementation of the retry feature in case of error from the packet forwarder (unable to transmit in case of TX already planned, unexpected error).

Due to the fact than this window is set at 1s after the RX1, I don't know if this feature must be implemented on the loraserver or on the lora-gateway-bridge to have a more fast answer/retry-mecanism.

The advantage of the RX2 is the ability to transmit with more power on the radio (in EU region).

Available to discuss and help around this functionnality.

Gilles

brocaar, 2018-10-10:
This has been implemented :)

```toml
  # RX window (Class-A).
  #
  # Set this to:
  # 0: RX1, fallback to RX2 (on RX1 scheduling error)
  # 1: RX1 only
  # 2: RX2 only
  rx_window=0
```

closed 2018-10-10 by brocaar

gillespilloudwyres, 2018-10-10:
nice job ! THanks !

commit 9bb40c40 (2018-10-10): Implement RX1 / RX2, RX1 and RX2 switch.

On RX1 / RX2 the frame will first be sent using RX1 parameters. On a
negative ACK, LoRa Server will retry using RX2 parameters.

Closes #314.