brocaar/chirpstack-network-server #318: Automatic Unwanted DataDown Packets (Moved from lora-app-server)

Shiipz, 2018-04-09:
###Is this a bug or a feature request?
Bug or Feature Change as of update.
Recently Upgraded the LoRa App Server, Now finding that every DataUp is generating an automatic DataDown packet addressed to devices from the gateway. Have not changed LoRa Device Code or Setup. All features in LoRa device are off including ADR, we are using UnconfirmedDataUp, ABP joining.
Anyone know where this setting is or what we are missing to turn off?

###What did you expect?
Expected No DownStream messages from gateway to device.

###What happened?
Have UnconfirmedDataDown message every upstream report with DataDown now having ADR true, (ADR still false in DataUp). Also RXParamSetupReq and RXTimingSetupReq in frame option field in DataDown. This has only started once we updated LoRa Server. FOpt is Null in DataUp

###What version are your using?
0.18.2 is our current version,
0.16.x was the previous version.

###How can your issue be reproduced?
Should be clear to see. If not maybe different configs are used. Trying to stop DataDown's.

--Update
We have filled in as many details (every field) on every device; Everything except Payload Codec. We do not wish the server or the gateway to be DataDown'ing the RXParamSetupReq and RXTimingSetupReq and this has only started since loraserver update. Anyway we can turn this off? We are working with the Loraserver.toml file config to see if there are any mismatches. 
Cheers.

brocaar, 2018-04-10:
When your devices does not respond to the `RXParamSetupReq` and `RXTimingSetupReq` mac-commands, then this is probably an issue with your device. The network-server tries to update the device configuration to keep it in sync with your `loraserver.toml` file. However it expects an acknowledgement from the device. As it does not receive such acknowledgement, it is resent.

Shiipz, 2018-04-10:
Thank you for this. It is what we are looking at.

We are using class C devices so I believe that it is meant to perform a Tx then two Rx Windows. We are trying to avoid extra radio communications (TX's) and would like to not need to send back an ack to the DataDown.

I realize that it is only once so will test with putting it in on first report. Interesting the DataDown is UnconfirmedDataDown that usually does not require an Ack.

Also, is this a feature of only the newest lora server (0.18.x+) ? we did not see it in the logs before (0.16.x) but was it still happening? What is making the loRaServer now think that the device needs synchronizing with this file? Is there a way for force it to believe that device and server are synchronized or to tell it to not perform the synchronization?

Thank you kindly, Also loving the work, this LoRaServer is amazing.

brocaar, 2018-04-15:
> Also, is this a feature of only the newest lora server (0.18.x+) ? we did not see it in the logs before (0.16.x) but was it still happening? What is making the loRaServer now think that the device [...]

In the latest versions, I've implemented features to keep the LoRa Server configuration in sync with the devices. E.g. when you change the RX2 frequency in the LoRa Server config, this new frequency will be automatically sent to your devices. So as long as your devices does not acknowledge these mac-commands, LoRa Server will retry, which is was you are seeing.

closed 2018-04-15 by brocaar