brocaar/chirpstack-network-server #320: Acknowledging the Server UnconfirmedDataDown Messages

Shiipz, 2018-04-20:
### Is this a bug or a feature request?
Bug / Request (Previous Issue)

### What did you expect?
1). If the server messages are ConfirmedDataDown, We expect devices to acknowledge them.
We are finding that the down messages are UnconfirmedDataDown, So devices will not acknowledge.

2). Also Gateways will not forward DataDown Messages.

### What happened?
1). DataDown messages that are auto generated by server with RX2 Frequency and Delay, continue as explained however we cannot acknowledge them with the RN2483 Modules at present. No command to force ack, and "AR" or auto response has no effect on this.

2). UDP messages to gateway appear slightly different in JSON format from TTN and This Lora Server,
Our three Lora Gateway Models do not transmit this LoRa server Gateway Messages, Is this a problem with the JSON UDP message format from server to gateway?

### What version are your using?
0.1.8

### How can your issue be reproduced?
Happens Every DataUp, (Can we have a checkbox maybe to prevent auto DataDown?)

### Could you share your log output?
for 2). captured with wireshark.

UDP packet from TTN: (works fine and device responds)
{"txpk":{
"imme":false,"tmst":755028988,"freq":869.525,"rfch":0,"powe":27,
"modu":"LORA","datr":"SF9BW125","codr":"4/5","ipol":true,"size":15,
"ncrc":true,"data":"YKMUASYABwABkmLyyx3A"}}

UDP packet from LoRa-Server: (Does not work, device does not show "mac_rx")
{"txpk":{
"imme":false,"tmst":737300180,"freq":867.1,"rfch":0,"powe":14,
"modu":"LORA","datr":"SF9BW125","codr":"4/5","ipol":true,"size":16,
"data":"YAMQoKyAEgAyd7rQ4YbSBQ==","brd":0,"ant":0}}

brocaar, 2018-04-23:
>  If the server messages are ConfirmedDataDown, We expect devices to acknowledge them.
We are finding that the down messages are UnconfirmedDataDown, So devices will not acknowledge.

When a downlink frame contains Mac-commands, it doesn't matter if it is using `ConfirmedDataDown` or `UnconfirmedDataDown`. The device will acknowledge the mac-commands, not the frame. Please see the LoRaWAN specification for more information how a device should acknowledge mac-commands.

> DataDown messages that are auto generated by server with RX2 Frequency and Delay
> "freq":867.1

Are you sure you configured 867.1 as RX2 frequency? It looks more like one of the extra channels configured by:

```
  # [[network_server.network_settings.extra_channels]]
  # frequency=867100000
  # min_dr=0
  # max_dr=5
```

> we cannot acknowledge them with the RN2483 Modules at present. No command to force ack, and "AR" or auto response has no effect on this.

With my RN2483 module, there is no need to do this by issuing commands. A mac-command is automatically ACKed (or nACKed) on the next uplink transmission. 

I've been testing with the 1.0.3 firmware for the RN2483. Are you on the same firmware version?

Shiipz, 2018-04-25:
Thank you for this.
We have re-read the specs and all seems clear. The MAC message will be auto responded to.

What we have reduced the problem to is that our gateways are not forwarding the UPD messages (part 2 of the questions above).
The format of the json UDP messages passed from LoRa Server to Gateway look almost identical to the TTN network (which down stream works and gateway transmits),
however
as mentioned the two parameters "brd" and "ant" are not in the TTN UDP Json messages but are in the LoRa Server Project Gateway UDP messages. This is the only difference we can see.

We are using Tektelic and Liard, not Multitech gateways. Would you know if this might be causing a problem that is preventing the gateways from transmitting??

We may also try to contact our gateway suppliers.

brocaar, 2018-04-30:
This has been fixed by Tektelic and I have this new firmware. I need to wrap up the last documentation bits, but it will be shared here: https://www.loraserver.io/lora-gateway-bridge/install/gateway/

brocaar, 2018-04-30:
Please see https://www.loraserver.io/lora-gateway-bridge/install/gateway/tektelic/ for the latest firmware.

closed 2018-04-30 by brocaar